prusa3d/PrusaSlicer #4577: Feature Request - Thumbnail generator uses filament color

cprasmu, 2020-07-26:
### Version
2.2.0

### Operating system type + version
Not Relavent

### 3D printer brand / version + firmware version (if known)
Prusa MK2.5S 

### Behavior
feature request  - When generating the thumbnail images it would be nice if the currently configured filament color was used.  An alternative would be the colour was keyed to the type of material ie PLA = RED, PETG = BLUE.

6ffm70, 2020-07-31:
One thought:

[image]
[image]

Visibility might suffer under variable contrast.

cprasmu, 2020-08-01:
> One thought:
> 
> [image]
> [image]
> 
> Visibility might suffer under variable contrast.

Yes see what you mean, I wonder if it could be possible for the edges get a contrasting highlight So even if the material colour was black or dark grey then the edges would still be visible... my other thought was that the colour would represent the material used and then in this case you would choose only colours that made sense.

th33xitus, 2020-11-19:
I would like to +1 to this FR.
I made a similar request back then to SuperSlicer and he kindly made it possible:
https://github.com/supermerill/SuperSlicer/issues/447

For Klipper users who use Mainsail or Fluidd as their Webinterface, those options could be super awesome.
Also there was an option to toggle rendering of the print surface on or off. So the issue mentioned above, that a gray cube on a gray plate looses contrast can be prevented.

I would really love to see such a feature in PrusaSlicer. Since its not only Prusa anymore to make use of those thumbnails :)

adolson, 2021-06-04:
+1 to this request. Would like to see it use the filament colour or the extruder colour, or even just a new option (I suggest "G-code thumnail color").

otto2704, 2021-08-28:
+1 too. would be very nice to configure thumbnail colour....

0x3333, 2022-03-18:
+1, SuperSlicer has this for some time. A PR would be accepted?

kukkjanos, 2023-01-26:
+1, must have function!

pl4nkt0n, 2023-02-27:
+1 thumb up from me

xidus90, 2023-03-20:
+1, thumb up

patrezp, 2023-03-25:
+1
But I really doubt that it will be implemented any time soon, as this FR is open for 3 years

dabohda, 2023-04-22:
+1
It would also be nice to disable the bed and print at the end of the gcode.

th33xitus, 2023-04-22:
So, seems like a lot of people are actually interested in a solution for that. It's kinda sad Prusa didn't even considered adding that feature over the last years, nor take the time to at least drop a short comment if that is something imaginable for the near future.

Anyways, this led to a small side-project in form of a post-processor i worked on together with a friend, nothing big but it does it's (limited) job. Maybe there are people interested in improving or just using it. It's currently not actively developed, as its state is "good enough for the effort that went in". So feel free to check it out and consider contributing to make it even better until Prusa may decide to support such functionality directly in PS: https://github.com/th33xitus/PrusaSlicerThumbnailPostProcessor

PS: No idea if there are any other projects out there that do a similar thing or work even better than mine, but in case they do, just drop a link in here and share them?

kubispe1, 2023-05-17:
Hi, with PS 2.6.0 beta1, thumbnails generated for G-code now use the same colors as in the 3D view, which means filament colors.
I am closing this issue. 
Thanks

psi-4ward, 2023-08-13:
[image]

[image]

[image]

Came here cause I wondered why Klipper shows the thumbs in grey. Anyone has a hint why theres only "default color"?

psi-4ward, 2023-08-14:
ahhh found it: There was a extruder-color set! :(

[image]

kb-, 2023-09-25:
Color should be set to the usual color when upgrading to 2.6.1.
It took me awhile to find out why my parts looked black on my new thumbnails! Extruder color was set to black...

daktylus, 2024-03-13:
+1 Referencing the Color to the Filament material should be easy