prusa3d/PrusaSlicer #4578: Slicer bug that fills in hole.

NeilMartin, 2020-07-26:
### Version
Version 2.2.0+Win64

### Operating system type + version
Windows 10 Home : Version	10.0.18362 Build 18362

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk2s (Stock) :  Firmware 3.2.3

### Behavior
BUG: When slicing a particular STL, the slicing solution incorrectly fills in a hole during a bridging layer.

Repo steps...
1) Open PrusaSlicer v2.2.0 in Win10
2) Click Add... and select testRails_02_05b.stl
3) Note that the default Print settings are "0.35mm FAST"
4) Select the Preview tab and scrub through the layers to find layer "25.75 (74)"
5) Notice that the bridging has filled in the hole...
[image]
, compare it to an older version of Prusa Slic3r that does not fill in the hole...
[image]
6) Further point of interest, if you now save the above problem in a 3MF file, shut down PrusaSlicer and then open that 3MF file, the slicer will now work and the hole is no longer filled (I found this out when writing this bug report, so it is required to perform steps 1->5 instead of loading the 3MF to repo the bug)...
[image]

#### Project File (.3MF) where problem occurs
testRails_02_05.3mf file (which wont show the problem)
testRails_02_05b.stl file (which will show the problem when following the above repo steps 1->5)
[Eurorack.zip](https://github.com/prusa3d/PrusaSlicer/files/4978557/Eurorack.zip)

rtyr, 2020-07-26:
Repairing the model should fix the issue.

NeilMartin, 2020-07-26:
The model was already repaired. The pre-repaired model completely fails and has many missing layers (but works fine in the old Prusa Slic3r)

rtyr, 2020-07-26:
Are you sure? Your STL doesn't look repaired. Here are my results (top: your STL/bottom: repaired). Same OS and PS version.

[image]

NeilMartin, 2020-07-26:
Here's my results before the repair. I've attached the un-repaired file, called testRails_02_05.stl

[image]

[testRails_02_05.zip](https://github.com/prusa3d/PrusaSlicer/files/4978839/testRails_02_05.zip)

How did you repair it? I used MeshMixer 3.5.474. I import, click on Analysis->Inspector->Auto Repair All, then I export.

NeilMartin, 2020-07-26:
Ah, I see how to repair testRails_02_05b.stl in PrusaSlicer now. However, PrusaSlicer does not seem to want to fix testRails_02_05.stl.
So I have to repair testRails_02_05.stl in MeshMixer to and export testRails_02_05b.stl, then use PrusaSlicer to fix testRails_02_05b.stl via Netfabb. Repairing in twice.

rtyr, 2020-07-27:
I don't recommend using Meshmixer for repairing models in general. It is better and easier to use 3DBuilder (integrated in Windows) or Netfabb application. I am using Meshmixer a lot to create multimaterial objects, but it is not very good when it comes to fixing models.

closed 2026-09-29 by prusaslicer-bot[bot]