prusa3d/PrusaSlicer #4579: Bug / Feature Request:  Double clicking STL files with custom data directory.

TheDuggem, 2020-07-26:
2.2.0+win64 and also code compiled from current master branch
Windows 10 Pro 64-bit 1909 build 18363.900

Double clicking STL files when associated with PS opens PS but the custom data directory cannot be specified.

Reproducing the issue:
Configuration files located in custom directory.
Confirm correct use of data directory when opening PS using --datadir location.
Associate STL files with PrusaSlicer.
Double-click on STL file.
Default data directory is used.

Attempted fix:
Associate STL files with .lnk shortcut file specifying --datadir location
Drag and drop STL file on shortcut, file opens using custom data directory
Double-clicking STL files still uses default directory

Feature Request:  Set data directory using environment variable.

foreachthing, 2020-07-27:
This is not a Slicer problem, it's a Windows (OS) issue which can be solved as following:
https://superuser.com/questions/361816/pass-command-line-arguments-to-windows-open-with

TheDuggem, 2020-07-28:
That's really useful and I had not come across that in my searching.

Agreed.  This is definitely not a slicer issue.  How is it handled in MacOS and Linux?

I'm not sure setting environment variables is any more accessible to the common user but it is safer than turning them loose on the registry.  I'm not sure either solution is optimal.

Thanks for this info!  I'll give it a try when I get a chance.

6ffm70, 2020-07-31:
**Background**
We have the same issue in Linux: we're sharing printers and materials, so we set up a common config dir in a Nextcloud folder (could also be an NFS, Samba or AFP or whatever mount) that is synced between all computers. As long as P/S is started from the shortcut containing the CLI-options, the correct config is used. When we open P/S by clicking a file in the file manager, the default settings in the local environment's config dir are used.

Same issue, but the same context of the Windows and likely the Mac circumstances. 

**Solution proposal**
While this could be solved on OS level, I could also imagine a symlink or .ink file referencing the central config directory in the installation directory of P/S (which we have tested on Win and Lin, but doesn't work).

foreachthing, 2020-08-03:
So, basically, PS should have an **option** to set the profile path.
This would solve all issues, if you don't want to use the default profile folder. This setting could then also be distributed to other computers.

TheDuggem, 2020-08-04:
By option do you mean a preference setting?  I think that's what you've been proposing in other comments.

My thoughts on a preferences setting vs environment variable:  

Using an environment variable is a 6-line change.  A few more lines if a separate variable for non-release builds is used.  It operates the same across all platforms.  The option isn't obvious to users but it doesn't look like a large number of users need it.  I think those who have a use case for this should generally know to read the documentation and know how to set an environment variable,  But we know how often that happens :)

Adding an entry to the preferences menu requires logic to handle the setting and all of the UI code that goes along with it.  It also requires translating prompts, tool tips, etc.  Preferences are stored in PrusaSlicer.ini which is located **in the data directory**.  To use a preference to specify the data directory now requires either having a copy of the .ini file in both the default location and the target directory or creating a new configuration file and associated code for this preference.

Documentation is needed for both implementations and both achieve the desired result but one has a much smaller impact.

foreachthing, 2020-08-04:
> Preferences are stored in PrusaSlicer.ini which is located in the data directory. 

Yeah, I didn't think about that :D
An env-var works, no question. It's just a pain to set in Windows. Maybe admin is required?

TheDuggem, 2020-09-05:
Solution provided by commit 683af51 

Thank you!

closed 2020-09-05 by TheDuggem