prusa3d/PrusaSlicer #4583: Drive Eject Button Causing Full Software Crash

GeekyFaye, 2020-07-27:
### Version
_Version 2.2.0 Win64_

### Operating system type + version
_Windows 10 (build 18363), 64-bit edition_

### 3D printer brand / version + firmware version (if known)
_Stock Prusa MK3S, Firmware 3.9.0_

### Behavior
* _After saving the gcode to the SD card, clicking the Eject button or hitting Ctrl+T results in the software going Non Responsive_
* _This happens 90% of the time with it randomly working rarely_
* _Happens regardless of STL or 3MF file, not file dependent_
* _Unsure how to reproduce this exactly as it does not always happen_

rtyr, 2021-02-23:
Is this issue still valid? Does it happen with 2.3.0?

GeekyFaye, 2021-02-23:
So, it's hinted at happening a few times since updating, however, I determined that I could fix the loop by simply initiating the Windows Safely Remove Media icon in the taskbar, which seems to imply that perhaps the software is unable to wake the drive out of hibernation perhaps? Most of the time it just works now, but whenever it looks like it's going to freeze and die, a click on that icon brings PrusaSlicer right back.

Perhaps something worth being aware of! But everything is okay now. 🙂 Thanks for following up!

rtyr, 2021-02-23:
Thanks for additional info.

closed 2026-09-29 by prusaslicer-bot[bot]