prusa3d/PrusaSlicer #4585: Top of part doubled ontop of part

Miksu80, 2020-07-28:
### Version
2.2.0

### Operating system type + version
Win 7 newest, 64bit

### 3D printer brand / version + firmware version (if known)
No matter, issue shows in slicer [Modded Tevo Tornado]

### Behavior
When slicing, for some reason slicer does extra top ontop of model on this .stl
[image]

[Kaiutinteline.zip](https://github.com/prusa3d/PrusaSlicer/files/4988048/Kaiutinteline.zip)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-07-28:
Disable "Detect thin walls" parameter.

Miksu80, 2020-07-29:
Thanks, that worked good.. I didnt know that this setting would do any harm to prints.

closed 2020-07-29 by Miksu80