prusa3d/PrusaSlicer #4586: Feature Request - Homogenous printing of bottom and top layer

wakalubatech, 2020-07-28:
### Version
PrusaSlicer Versin 2.2.0, all OS, using MK3s with MMU2 and latest Firmware

### Problem
The current software does not allow settings for linear printing of the bottom and top layers with homogeneous feed.
If the ground is printed, the current optimization is to get the shortest possible extruder paths. This is a completely useless optimization because it significantly worsens the surface quality. During printing, gaps repeatedly appear in the middle of the printing area, e.g. through holes that are then closed from two sides. If you look at the printing surface against the light, this results in an unacceptable picture, since the material is melted onto the existing material in different directions.

### Solution
One remedy is a completely monotonous painting over the printing surface, so that printing is only ever carried out on one side of the already printed material.
The direction should also be selectable with x or y to increase the homogeneity, since the holes could be aligned in one direction.

n8bot, 2020-07-28:
> [...] layers with homogeneous feed. [...]

> [...] completely **[monotonous](https://github.com/prusa3d/PrusaSlicer/blob/06a8bfa5887c2e1871d3e17c9b46b46433caf210/src/libslic3r/Fill/FillBase.hpp#L40)** painting [...] [emphasis mine]

If only there was some feature in PrusaSlicer 2.3.0 alpha0, that could do this. We could call it homogeneous infill!! Or, something!

closed 2020-07-29 by rtyr

timur-tabi, 2020-10-17:
For the record, "monotonous" is not the right word. The correct word is "monotonic".

rtyr, 2020-10-17:
It is already renamed.

[image]