prusa3d/PrusaSlicer #4588: Feature request: Modifier setting for infill angle

terribleperson, 2020-07-29:
### Version
2.2.0

When you add a modifier shape, one of the setting categories available is Infill. In that category you can select to change 'Fill density' and 'Fill pattern'. I suggest making the 'Print Settings > Infill > Advanced > Fill angle' setting controllable as a modifier setting in the infill category. 

One of the reasons to use multiple infill types or densities within a print is for parts that are shaped substantially different at different layers or regions that experience substantially different forces. As an example I will show this shelf bracket I am working on. 
[image]
For the round and largely vertical sections of the bracket I am using gyroid infill. It is my go-to infill for its ease of printing and relatively isotropic behavior. For long, thin structures it might be preferable to use an infill that will fill the space with continuous extrusions. I'm doing that here by setting the 'Fill pattern' to 'grid' in a 'Height range Modifier'.
[image]
[image]
In the case of this design I am lucky that both sections I wanted to fill with grid infill are oriented in roughly the same direction. I was able to adjust the profile 'Fill angle' setting to get the result I wanted in both sections. It would be better to be able to set 'Fill angle' as a modifier setting alongside 'Fill pattern' to be able to adjust infill to fit different situations.

neophyl, 2020-07-29:
Erm you can already do that.  Fill Angle is already one of the options.
[image]

Instead of just selecting Infill, select Add Settings.  That gives you access to all the available options and not just the commonly used ones that the quick options give you.

rtyr, 2020-07-29:
Yes, it's like @neophyl described.

[image]
[image]

Closing.

closed 2020-07-29 by rtyr