prusa3d/PrusaSlicer #4590: Wipe tower should respect the per-layer speed settings

redlukas, 2020-07-29:
### Version
2.2.0

### Operating system type + version
Windows 10 2004

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3s + MMU2s, Fimware Version 3.9.0

### Behavior
_Problem description_
* When a custom first layer speed is set, the first layer of the wipe tower gets printed  at the default speed
* _Steps needed to reproduce the problem_

1. Load any multimaterial model into PrusaSlicer with the Printer set to MMU printing

2. Go to Print Settings -> Speed -> Modifiers -> First layer speed and enter a custom first layer speed (I use 20%)

3. export and print the sliced model

* _Expected Results_
The first layer of the wipe tower should be printed at the same custom speed as the first layer of the model
* _Actual Results_
  the first layer of the wipe tower will be printed at solid infill speed, while the first

 layer of your model will be printed at the custom speed

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Sheep.zip](https://github.com/prusa3d/PrusaSlicer/files/4993848/Sheep.zip)

redlukas, 2020-07-30:
Just a short addition: this does not pertain to the whole first layer of the wipe tower, only the part that gets deposited before depositing the first layer of the object. if the object has a color change in the first layer, PrusaSlicer will only print the outline of the wipe tower in the erroneous default speed, print the first color of the object (at the custom first layer speed) and then continue on the wipe tower (also at the custom first layer speed).

tommylux, 2022-11-25:
[image]
My wipe tower prints too fast on the first layer, it doesn't have chance to stick to the bed.

lukasmatena, 2023-09-05:
Fixed in [2.6.0](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0). Closing.

closed 2023-09-05 by lukasmatena