prusa3d/PrusaSlicer #4593: M117 progress and remaining time as option for M73

KimmoHop, 2020-07-30:
### Version
2.2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega

### Behavior
Printer settings "Supports remaining times" exports progress (0..100%) and remaining time (minutes) using M73 gcode.
Marling supports M73, but not from SD card (at least they say so) and it supports only progress (P), not remaining time.

M73 can be postprocessed to M117, but with Windows PowerShell it (due to going through full gcode file) takes ~ 1 minute / printing hour, so it is very slow! Adding M73 commands don't seem to take much time.

It would be better to have option to print progress and remaining time using M117 command. Nothing fancy, "  6%  58m" would do fine.
An alternative could be inbuilt postprocessing for M73 -> M117 conversion.

This is sort of feature request, or "good behaviour request" to support non-Prusa printers.

scheric, 2020-08-10:
M73 P[xxx] R[xxx]  can be used with Marlin. Print time only works with graphical displays:

https://github.com/MarlinFirmware/Marlin/blob/fd9c6983d30f70171d9d9a9c55962689fa4cdd85/Marlin/Configuration_adv.h#L1083-L1111

> M73 can be postprocessed to M117, but with Windows PowerShell it (due to going through full gcode file) takes ~ 1 minute / printing hour, so it is very slow! Adding M73 commands don't seem to take much time.

Python can proses my 6h print in 150 mS to 200 mS. (Depending on if everything needs to be printed in the console.)

> An alternative could be inbuilt postprocessing for M73 -> M117 conversion.

Yes, an extra checkbox would be the best solution.

KimmoHop, 2020-08-11:
I made a test to see how it could be done: [initial version](https://github.com/KimmoHop/PrusaSlicer/commit/05aee1230251dc2204250d687035983a88180261) and [small fix](https://github.com/KimmoHop/PrusaSlicer/commit/e0d7cd516ad2788e5fb5d6bb18d5d346c149e7dc)

My graphical (Marlin) LCD already shows progress bar (probably based on how much bytes of the gcode file has been read) but it is not accurate. Also estimated print time can be added to file name, so it will be shown on display (if scroll long filenames in Marlin is enabled) and that can be compared to work time. Anycubic i3 Mega by default shows progress bar and work time, but no long names. I don't know how e.g. Ender 3 works.

Anyway, this is better, faster and less work for users when done in slicer. Building and updating printer FW or installing Python and finding correct script is not so simple.

Alternative could be, if postprocessing accepted regexp rule or added regexp/C format rules for source and target, little like in ideaMaker (though it may not support parsing variables):
[image]

Postprocessing would be more generic, but not so easily found option.

scheric, 2020-08-15:
> I made a test to see how it could be done: initial version and small fix

Perhaps make a PR from it. 

> Anyway, this is better, faster and less work for users when done in slicer.

Yes, but only if the PR is successful or precompiled.

> installing Python and finding correct script is not so simple.

My script is on GitHub: https://github.com/scheric/prusaslicer-m73-macro

Hades32, 2020-08-16:
> I made a test to see how it could be done: [initial version](https://github.com/KimmoHop/PrusaSlicer/commit/05aee1230251dc2204250d687035983a88180261) and [small fix](https://github.com/KimmoHop/PrusaSlicer/commit/e0d7cd516ad2788e5fb5d6bb18d5d346c149e7dc)
> 
Cool, a small tip: Don't work on master, but another branch,  when forking a repo. This makes PRs so much nicer and makes it easier for others to find the changes 👍

Hades32, 2020-08-16:
> My script is on GitHub: https://github.com/scheric/prusaslicer-m73-macro

The easiest way I found, with a simple per one-liner (is e.g. included with Git for Windows)
```
perl -i.bak -ne 'if ($_ =~/M73 P([0-9]+) R([0-9]+)/){ ($pct,$min)=($1,$2); $h=int($min/60); $min=$min-60*$h; print "M114 $pct % - $h:$min\n"; } else { pri
nt; }' /c/Temp/thinker_0.2mm.gcode
```
This will also create a backup of the original file.

KimmoHop, 2020-08-18:
I'll adapt to a branch and will do PR.
I don't think Windows comes with Python nor Perl (thank goodness ;D) - so if they do not come as part of the slicer, they are an extra hop. But multipurpose ;)

KimmoHop, 2022-02-04:
With coming 2.4.1 converting M73 to M117 should be possible without changes in slicer or additional software, though only in minutes ;)

bubnikv, 2023-02-07:
You may use built in regexp substitutions in PrusaSlicer for this purpose.

KimmoHop, 2023-02-08:
> You may use built in regexp substitutions in PrusaSlicer for this purpose.

Isn't it so, that M73 usually has only P or R (when each changes) -> how does regexp combine (remember) them to M117 with both?

I hope correct regexp comes prebuilt in slicer (dropdown etc) with 1-click usage. 99% of users won't know how to create correct regex...

closed 2026-09-29 by prusaslicer-bot[bot]