prusa3d/PrusaSlicer #4595: saving a config on a mac remove all the print setting

BenoitDesrosiers, 2020-07-31:
### Version
_Version of PrusaSlicer used goes here_versions_2.1.1+

_Use `About->About PrusaSlicer` for release 

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version 
mac os 10.13.6

### Behavior

* _Describe the problem_
If I take a configuration snapshot, all the option in the Print settings menu disappear.  
I would say that in general, the configuration snapshot just don't work on mac. I didn't try on pc, but on mac, it is useless.

rtyr, 2021-02-23:
Is this issue still valid? Does it happen with 2.3.0?

rtyr, 2023-09-12:
No response. I am closing the issue.

closed 2023-09-12 by rtyr