prusa3d/PrusaSlicer #4598: Feature Request: Please remove unrealistic precision from filament consumption prediction

mild-lemon, 2020-08-02:
### Version
Version: 2.2.0+
Build: PrusaSlicer-2.2.0+-202003211132

### Operating system type + version
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS Catalina Version 10.15.6 (Build 19G73)

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro
fw-version 1.1.8

### Behavior
* _Describe the problem_
Unnecessary precision in filament consumption prediction.
In comparison, estimated printing time is not given in 1/100th of a second either, but with a reasonable minute precision only.

* _Steps needed to reproduce the problem_
1. Set "Output Filename Format" to include `..._{total_weight}g_...`
2. Slice any file.

* _Expected Results_
`..._62.7g_...` in output filename, or
`..._63g_...` in output filename

* _Actual Results_
`..._62.6601g_...` in output filename

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  included.

_Is this a new feature request?_
YES

#### Project File (.3MF) where problem occurs
any file
[image]

n8bot, 2020-08-02:
You can cast the value to an integer using the macro function int(). If you wanna round up, add 0.5. This will output no decimal places.

mild-lemon, 2020-08-02:
* _Steps needed to resolve the problem_
1. Set "Output Filename Format" to include `...__{int((total_weight+0.5))}g_...`
2. Slice any file.

* _Expected Results_
`..._62.7g_...` in output filename, or
`..._63g_...` in output filename

* _Actual Results_
`..._63g_...` in output filename

==> SOLVED!

closed 2020-08-02 by mild-lemon