prusa3d/PrusaSlicer #4599: PrusaSlicer crashes if Windows Key is pressed during Rotate

anttikarhu, 2020-08-02:
### Version
2.2.0+win64

### Operating system type + version
Windows 10, 64 bit

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
PrusaSlicer crashes if Windows Key is pressed during Rotate.

How to reproduce:
* Open PrusaSlicer
* Drop a model to build platter (Benchy will do nicely)
* Select rotate tool
* Grab a rotate handle with mouse (no matter which one) and move mouse, don't let go of the mouse button
* Press Windows Key 

-> Pruca Slicer crashes.

"A debugging check in this application has failed.

..\..\src\common\wincmn.cpp(3375): assert ""Assert failure" failed in
DoNotifyWindowsAboutCaptureLost(): window that captured the mouse didn't process
wxEVT_MOUSE_CAPTURE_LOST"

Noticed this issue when I rotated, and tried to snap the rotate to 90 degrees by pressing Shift, Ctrl, Alt and accidently Windows Key :) 

#### Project File (.3MF) where problem occurs
N/A

anttikarhu, 2020-08-02:
[image]

n8bot, 2020-08-02:
I'm pretty sure this is already fixed for the next version. It's not present in current builds of the master branch.

rtyr, 2020-08-02:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4404. Closing.

closed 2020-08-02 by rtyr