prusa3d/PrusaSlicer #4600: Sliced object doesn't look like loaded STL object

mobil750, 2020-08-02:
### Version
PrusaSlicer 2.2.0+

### Operating system type + version
Mac OS Catalina and all updates

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S with latest firmware

### Behavior
* I have an object loaded from Thingiverse with a few small adaptions. It is constructed via openSCAD and looks within openSCAD as expected. Same when I import the STL into PrusaSlicer. But after slicing some holes are missing and the object  doesn't look like the STL one.  The original object from Thingiverse behaves the same - slicing is incorrect.

* _Steps needed to reproduce the problem_
1. Start PrusaSlicer
2. import the attached STL into PrusaSlicer -> looks as expected and designed
3. Slice it --> result is missing the main hole in the middle and a small hole on one side.
4. Print is as sliced screenshot missing relevant holes

* _Expected Results_
Sliced object should show the designed object with all details (holes)

* screenshot of imported STL and of the sliced result are attached.

Further you find the 3MF file and the STL file (tray-test.stl) as zipped files.

_Is this a new feature request?_
No

[tray-test.3MF.zip](https://github.com/prusa3d/PrusaSlicer/files/5012567/tray-test.3MF.zip)

[tray-test.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/5012570/tray-test.stl.zip)

[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2020-08-02:
Openscad is notorious for these sorts of issues.  Have you tried the fix through netfabb option ?

mobil750, 2020-08-04:
It seems that the problem is typically for openSCAD when using "big" numbers for detailed cylinders (%fn=360). I was able to change the code to get an acceptable result.

Thanks

rtyr, 2020-08-15:
Can we close this issue?

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo