prusa3d/PrusaSlicer #4603: Retraction performed before print starts

gnydick, 2020-08-03:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 64-bit version 2004

### 3D printer brand / version + firmware version (if known)
RailCore II ZLT - RepRapFirmware 3.1.1

### Scenario
* dual extruder
* two priming extrusions
* two different nozzle sizes

### Behavior
You can see after the `; end of start code`, which denotes the end of the custom start g-code, the generated code begins.  When you reach the first layer, a **change layer** is performed in the slicer logic, denoted by `<----- HERE`,  so an extrusion is performed, which shouldn't happen.

#### Beginning of g-code
```
; generated by PrusaSlicer 2.2.0+win64 on 2020-08-03 at 19:04:46 UTC

; 

; external perimeters extrusion width = 0.68mm
; perimeters extrusion width = 0.68mm
; infill extrusion width = 0.68mm
; solid infill extrusion width = 0.68mm
; top infill extrusion width = 0.60mm

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm

M107 ; disable fan
M290 R0 S0 ; clear babystepping
G29 S2 ; clear heightmap
M561 ; disable bed compensation

;G28 ; home all axes
;G32 ; Autocalibrate bed
;G28 Z; home just Z
;G29 S1 load heightmap

M140 S70
M104 S215 T0; set 1st nozzle heater to first layer temperature
M104 S215 T1; set 1st nozzle heater to first layer temperature

M190 S70
M109 S215 T0; set 1st nozzle heater to first layer temperature
M109 S215 T1; set 1st nozzle heater to first layer temperature

T0 ; prime T0
G1 X30.0 Y1.0 Z0.3 F1000.0 ; prepare to prime
G92 E0 ; reset extrusion distance

G91 ; relative positioning
G1 Y30.0 E12.5 F1000.0 Z-0.1 ; priming
G1 Y65.0 E9.0 Z-0.2 F1000.0 ; priming

;wipe T0
G1 Y75 F6000
G1 Y-10 F6000
G1 Y10 F6000

G90 ; absolute positioning

G1 Z2 F1000.0

T1 ; prime T1
G1 X31.0 Y0.0 Z0.3 F1000.0 ; prepare to prime
G92 E0 ; reset extrusion distance

G91 ; relative positioning
G1 X30.0 E12.5 F1000.0 Z-0.1 ; priming
G1 X65.0 E9.0 Z-0.2 F1000.0 ; priming

;wipe T1
G1 X75 F6000
G1 X-10 F6000
G1 X10 F6000

G90 ; absolute positioning

G1 Z2 F1000.0

; end of start code
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
T0 ; change extruder
M900 K0.04 ; Filament gcode LA 1.5
M900 K18 ; Filament gcode LA 1.0
;BEFORE_LAYER_CHANGE                    <--------------- HERE
;0.2

G1 E-1.00000 F2400.00000 ; retract
G1 Z0.600 F7800.000 ; lift Z
;AFTER_LAYER_CHANGE
;0.2
```

n8bot, 2020-08-03:
It would be so nice to be able to customize the GCode output like this, and a lot more: suppress certain gcodes; modify the actual GCode itself and the parameter letters (words) it uses, etc. I realize this isn't something Prusa devs would be interested in developing or supporting. I wonder in what way the features could be implemented unobtrusively for power users, while not affecting Prusa customers?

gnydick, 2020-08-03:
I think the problem is the logic is conflating the notions of "do before"
and "do before change".

"Do before" would perform some g-code before even the first layer.

"Do before change" should only perform some g-code starting at the second
layer.

lukasmatena, 2020-08-04:
@gnydick The retraction has nothing to do with before/after layer custom gcodes. It is performed because 'Retract on layer change' is turned on in Printer Settings. Turn it off if you don't like it.

Apart from that, why is the retraction such a problem? If it wasn't there, you'd still get a retraction immediately after that layer change, because the layer will start with a travel move. What difference does it make?

gnydick, 2020-08-04:
There is no layer change at the beginning of a print.

Almost everyone has a priming function in the beginning to, well, prime the
hot end. Retracting immediately after priming before the print actually
starts defeats the purpose of priming.

gnydick, 2020-08-04:
And, btw, I know it has nothing to do with the before custom g-code. I
didn't say it did. I was saying the logic was conflated.

"On layer change" should only trigger when there is a layer change. Since
there is no layer 0, why am I getting a reaction?

The reason I said it's conflated is empirical. If you look at the g-code
output, the retraction is placed between the BEFORE and AFTER markers.
Again, even without retraction enabled, I wouldn't want my BEFORE or AFTER
LAYER CHANGE scripts to execute before the first layer even prints, because
there has been no layer change. This is a defect.

closed 2026-09-29 by prusaslicer-bot[bot]