prusa3d/PrusaSlicer #4604: View resets to "Color Print" every time after slicing

geuis, 2020-08-05:
### Version
2.20 win10

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
i3 MK3S

### Behavior
I don't know why but recently when I slice my models they're being sliced as mutli color. I'm not seeing any options to toggle this on or off. Its causing my prints to pause multiple times and I have to manually remove and replace the filament. 

I think this is related to "Color Print". This isn't an option I've ever used.

How do I turn this feature off?

rtyr, 2020-08-15:
You most likely added color change accidentally in the preview. See the image below. 

Can we close this issue?

[image]

geuis, 2020-09-03:
No, I didn’t add it on accident. This issue has been happening on multiple new prints.

rtyr, 2020-09-03:
Please make a screenshot of the whole PrusaSlicer window when you have this issue and save your project to 3mf file.

rtyr, 2020-09-24:
No response, no other reports of this issue. I still think that color change was added by accident or saved to 3MF project by accident. Closing. Feel free to reopen if it happens again and you will have some more details to share.

closed 2020-09-24 by rtyr