prusa3d/PrusaSlicer #4606: UX/UI: Revise warning "Some objects are too close; your extruder will collide with them."

combs, 2020-08-06:
### Version
_Version of PrusaSlicer used goes here_
2.2.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
linux-x64 and OS X 10.13

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa i3 Mk3, upgraded to Mk3S

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

When slicing multiple objects with "sequential printing" ("complete individual objects"), if an object is too close to another object, PrusaSlicer pops a modal: "Some objects are too close; your extruder will collide with them."

[image]

This modal pops open again every time an object is moved, added, or otherwise altered. This can occur dozens of times if you are trying to arrange a few objects on the bed. 

When an object is detected outside the working area (a similar condition that prevents slicing), instead of a modal, the a red warning is superimposed on the 3d view and a non-modal warning is shown.

[image]

Similarly, after slicing, if a toolpath is outside the working area, the 3d view background becomes red and a warning is superimposed on the 3d view; no modal is shown.

[image]

In this issue I propose to change the "objects too close" warning from a modal to the "outside working area" or "toolpath" style, which allows users to continue working without clicking OK a lot.

_Is this a new feature request?_
Improvement to existing feature

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[Too-close.zip](https://github.com/prusa3d/PrusaSlicer/files/5035315/Too-close.zip)

darigovresearch, 2020-09-06:
Nice work @combs for documenting this issue! We think that in general this should be an easy fix as it requires just switching out the function so that the error text is displayed not in the modal but C++ is not our specialty.

Relevant files appear to be 
- [src/libslic3r/Print.cpp](https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/Print.cpp#L1231)
- [src/slic3r/GUI/GLCanvas3D.cpp](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/GLCanvas3D.cpp#L631)
- [src/slic3r/GUI/GLCanvas3D.hpp](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/GLCanvas3D.hpp#L321)

Maybe a maintainer could tag this as a good first issue to help increase the likelihood that someone who knows C++ could resolve it

lukasmatena, 2020-09-06:
I think this issue is no longer relevant after recent changes (behold in `master`). The modal dialog was replaced by non-modal notification in the bottom right corner:

[image]

In fact, I'm closing the issue as resolved. I hope everyone agrees.

closed 2020-09-06 by lukasmatena