prusa3d/PrusaSlicer #4607: Feature Request: Additional filename placeholders

therealzanfar, 2020-08-06:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10 Pro x64, ver. 2004, build 19041.388, feature pack 120.2202.130.0

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s stock

### Feature Request

Additional filename placeholders for use in Print Settings -> Output Options -> Output filename format

* Parent Folder

    My workflow commonly involves multiple STLs per project, all saved in the same folder. These parts will commonly have similar or identical names as parts in other project--for example, multiple projects may include a similar, but not identical, "axle.stl". While the filename is sufficient to identify the part in a folder hierarchy, when exported, PrusaSlicer defaults to naming the GCode file "axle.gcode" which gets lost in the printers single-level storage.

    **Please add a `{parent_folder}` placeholder** that matches the name of the immediate ancestor folder of the primary object file.

* Object Count

    My workflow commonly involves generating multiple GCode files from a single STL object usually in different quantities. That is, I might have a GCode file with 1, 4, and 16 duplicate objects on the build plate. Which file gets printed depends on the quantity of parts needed--sometimes I'm doing a production run and I use the full-plate x16 version, while sometimes I just need a quick replacement or a prototype and only want a single object.

    **Please add a `{object_count}` placeholder** that is the count of objects on the build plate.

Together, this means my ideal Output filename format would be something like:

    {parent_folder}_{input_filename_base}x{object_count}_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}.gcode

josedpedroso, 2020-08-14:
I also have some new placeholder variables I'd like to request:

- **output_filename**: the idea would be to build print progress messages with M117 and conditional g-code, to show different stats on odd or even layers, for instance, but always maintaining the real filename visible.

- **layer_min_volumetric_rate**, **layer_max_volumetric_rate**, **layer_avg_volumetric_rate**: I was thinking about making some sort of automatic temperature adjustment script on layer changes, like Marlin's autotemp feature, but steadier.

sslupsky, 2020-08-26:
I like the suggestion of the object count placeholder.  This would be very useful.

carlosjln, 2023-09-20:
Well this is still hanging, but I'd like to have the stl file name not the project filename. I save the project with specific print settings and just swap the files but I'd like to export with the sliced STL filename.

clintonefisher, 2024-10-26:
Hi everyone,
I would like to also see: **{filaments_used}**, or **{tools_used}**, that would output **"F"**, followed by a list of filaments/tools used, ie: ;"1-5.
If a print uses filament/tool 3 in a print, the output would be **"F3"**.  Filaments 2 & 3 would be **"F23"**. 
All of my MMU2 filaments in use would be **"F12345"**.
This would help MMU users to know what filaments to use based on the filename.
(It does break at 10 tools, so how to deal with that?   Hmm.  Underscores or dashes for all tools in output string if a print has >= 10 tools.  That works. I.e.: **"F1_3_5_16_19"**.  
What do others think about this idea?  I would find it super-handy, as I have to do it manually now.

closed 2026-09-29 by prusaslicer-bot[bot]