prusa3d/PrusaSlicer #4608: Request: strength-optimized “combine infill every…” feature

fiveangle, 2020-08-06:
The current implementation of “combined infill every…“ is designed to speed printing by combining multiple infill layers, reducing infill print paths to 1/η where η is the integer number of combined layers the user selects, typically 1-3.

My proposal is to extend this feature to optimize intra-layer adhesion for functional models where infill paths interface adjacent infill paths which occurs when infill density is extremely high, such as 90-100% (eg near-solid infill).

This increased intra-layer strength￼ is achieved by alternating the combined infill pattern so that they print staggered across the infill paths, allowing the combined infill layers to adhere to each other not only on the z layer boundary, but along the boundary of adjacent alternating combined layers, resulting in a “brick lay” effect in the z axis direction, dramatically increasing intra-layer adhesion and torque withstanding capability.

As this feature is likely only useful for high infill where infill paths touch adjacent infill paths, implementing this feature only for infill patterns where it makes sense to do so such as Lines, Rectilinear, Wave, Octagon Spiral, Concentric, etc. may ease implementation effort by eliminating possible edge cases for more complex infill patterns where near-solid infill does not make functional sense (3D Hex, Gyroid, etc.)

Below is an example of a 9-layer in total print with 1 top and bottom layer with the simplest “Line” infill with the proposed change:

[image]

faboaic, 2020-08-08:
Absolutely great as well as simple idea. 
I hope that this will get implemented.

mepoland, 2021-12-12:
This would be great for a thin rod aligned with the Z axis that must undergo torque in use. Currently the part breaks consistently between layers because they can't handle the shear force. I wonder if this idea could be extended to perimeters as well, since that's where it would yield the biggest improvement in my torque scenario.

closed 2026-09-29 by prusaslicer-bot[bot]