prusa3d/PrusaSlicer #4609: Failed processing of the output_filename_format template.

bheiland, 2020-08-07:
### Version
2.3.0 alpha 0 code viewer version

bcd998f9f1169f8e41e29f7615a0f4fe5d887014
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Mac 10.15.5

### 3D printer brand / version + firmware version (if known)
prusa mk3s

### Behavior
Add stl file, and then slice file. interface works properly until I try to Export G-Code
[image]

enricoturri1966, 2020-08-07:
This issue should be fixed by: 64001c0fe5794dd747cb388b302cb8397e741ecc

bheiland, 2020-08-08:
Fixed in 64001c0 ,   tested fix and works properly now ,  Thanks!

closed 2020-08-08 by bheiland