prusa3d/PrusaSlicer #4610: Error on Startup

rocknrude, 2020-08-07:
### Version
2.2.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Win 10 64bit Home

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Behavior
App stopped working
Clicking on app results in this error. I have uninstalled and redownloaded program but still get same error

_
[image]

n8bot, 2020-08-07:
Have you tried deleting the file listed in the error message, as it says to do?

rocknrude, 2020-08-07:
Yes. Did that and it fixed it. Thank you.

bgiot, 2020-08-07:
@rocknrude  Can you close the issue? Thank you!

lukasmatena, 2020-08-07:
@rocknrude Don't you still have the faulty ini file somewhere in trash, so we can investigate how the problem started in the first place?

ntmaster26, 2020-08-07:
Not sure but I’m locking up all the time ... not responding

closed 2020-08-07 by rocknrude