prusa3d/PrusaSlicer #4612: Separated top layer when 0 infill

Pitel, 2020-08-08:
### Version
PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage

### Operating system type + version
Ubuntu

### 3D printer brand / version + firmware version (if known)
Prusa MINI 4.2.0

### Behavior
I have this very simple box. When I slice with 0 infill, the top layer below the walls gets separated. When I use more infill, it's one continous layer.

I suppose the continous infill is better.

[image]

[image]

#### Project File (.3MF) where problem occurs
[box.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/5044809/box.3mf.gz)

JohnnyDeer, 2020-08-11:
Hi @Pitel, I can confirm this. Thank you for reporting.

Pitel, 2020-10-18:
Still happening in 2.3.0-alpha1.

closed 2026-09-29 by prusaslicer-bot[bot]