prusa3d/PrusaSlicer #4613: Display material used by support structures seperately

dirrgang, 2020-08-08:
While playing around with the prusaslicer (my printer hasn't arrived yet), I toyed with a few different orientations of the object, and the support structures it'd require to print properly.

I noticed there seems to be no easy way to gauge the amount of support material used in a sliced object. To me it seems like this would be a good way to gauge the "effectiveness" of the 3D print object's orientation. In general it'd seem the fewer the better.

It seems like the amount of filament used on the different parts of the print (infill, support etc.) would be very useful, or at least interesting.

My experience in C/C++ is limited, but I wouldn't mind trying to add it myself. At the moment it seems like any extrusion is just added to a tachometer variable, which is used to make the calculations in the end.

Any suggestions where/how to implement this feature? First idea would be possibly extending the extrude function with a variable that declares the type of component the extruded filament is used for, so it can be stored in the total tachometer, as well as new tachometers for the other types of filament. But that might not be the best place for it to be implemented... any suggestions?

b-irwin93, 2020-10-18:
I have been thinking about this also. Perhaps having it in brackets after the Used Filament i.e. Used Filament (g) : 19.09 [3.49 support].

Even having different options for this such as including the totals for the model or a breakdown for individual models.

I don't have much coding experience so I have no hope adding this myself.

lukasmatena, 2020-10-19:
> I noticed there seems to be no easy way to gauge the amount of support material used in a sliced object.

There is not, but the new 2.3.0-alpha1 is able to show time estimate per feature. This can be now used as at least *some* measure of how much are there.

> Perhaps having it in brackets after the Used Filament i.e. Used Filament (g) : 19.09 [3.49 support].

bubnikv, 2021-09-22:
Implemented, certainly in PrusaSlicer 2.4.0-alpha1.
[image]

closed 2021-09-22 by bubnikv