prusa3d/PrusaSlicer #4618: Feature Request: Layer time / Extruder temp modifier factor

clytle374, 2020-08-10:
I came up with this idea due to a layer separation issue on large abs prints. My temp was tested on a small model, when switching to a larger model I do not get good layer adhesion.  Some testing shows that a small part, with this ABS, need a temp of 210 to prevent part from being to hot and deforming. A large model needs much higher temp of around 255 to get good layer adhesion as the previous layer has greatly cooled.   The problem is when a part starts large and tappers small at the top.  Using the part fan on ABS seems to often really increase warping. 

My proposal is to add something like the part cooling options currently provided(fan if layer time is below, slow down if layer time below) and add min and max temperatures for the material with a threshold for layer time as it takes less heat to fuse to a already hot layer.   It could also be tied to volume of a protrusion for parts with multiple small protrusions on the top, but that would be more difficult. 

Just a thought.

BeatSlayer, 2020-08-10:
The best solution to your problem it's a heated chamber.

clytle374, 2020-08-11:
Best solution is injection molding.  Heated chamber would help.  Just a suggestion.

josedpedroso, 2020-08-14:
This could be done with some conditional g-code, if the **layer_time**variable existed.

rtyr, 2021-02-24:
I think "Enable fan if layer time is below" option in combination with very low fan speed (for example 15-20%) should be enough to solve this.

closed 2026-09-29 by prusaslicer-bot[bot]