prusa3d/PrusaSlicer #4621: SL1  Mixed platter of non & supported objects elevations

nionio6915, 2020-08-10:
### Version
2.2.0_win64

### Operating system type + version
win64, 1909
### 3D printer brand / version + firmware version (if known)
SL1, but does not matter

### Behavior
* _Describe the problem_
When exporting a mixed plate to STL with supports, objects on a platter that are on the build plate are raised to the elavation for supports, and ones that are supported remain on the build plate. 

* _Steps needed to reproduce the problem_
Add object, orientate it to print from the build plate. 
Add another object and assign supports to it. 
Both objects appear correct on the PS platter preview. 
Save the plate as STL with supports. 

* _Expected Results_
All elements on the buildplate. 

* _Actual Results_
The result is the objects are at different elevations. 

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
No. 

#### Project File (.3MF) where problem occurs
[test-combined.zip](https://github.com/prusa3d/PrusaSlicer/files/5052790/test-combined.zip)

tamasmeszaros, 2020-08-17:
Hello!

While investigating your 3mf, I've noticed that the cylinder is not configured to be printed from the build plate. Its configured to have supports and pad as usual. This way, its normal that the elevation remains. What is strange that you have no supports for the cylinder on your pictures. Something went wrong in your case that is for sure. If you disable the pad and supports for the cylinder individually, slice the scene and then export to stl, the cylinder should have no elevation in the exported stl.

bubnikv, 2021-09-22:
I see you are using PrusaSlicer to generate STLs for 3rd party SLA printers. You likely placed some support points at the side or at the top of the cylinder, thus PrusaSlicer tried to generate support structures for these points and it failed, however it lifted the object. If you tried to slice such an object in PrusaSlicer, it would fail. In PrusaSlicer 2.4.0-alpha1 we newly show a notification that something went wrong when trying to calculate the support legs.

We believe the current behavior is sufficient.

closed 2021-09-22 by bubnikv

nionio6915, 2021-09-22:
this reply is disappointing. i am using it for both SL1 and non-prusa printers. 

the issue still remains. If you arrange a plate of mixed objects, some on supports and some not on supports, and try to export it, the non-supported objects are rasied. 

please reconsider closing this issue.

lukasmatena, 2021-09-23:
@nionio6915 I just checked and while I indeed reproduced the problem in 2.2.0, it seems to be fixed in 2.4.0-alpha1. Can you please test that? Thanks.