prusa3d/PrusaSlicer #4624: Tiny perimeters happening at start of print leading to failures

morphias2004, 2020-08-11:
### Version

2.2.0+win64

### Operating system type + version

Windows 10 Pro x64 1909 OS Build 18363.959 with latest Microsoft updates

### 3D printer brand / version + firmware version (if known)

3.9.0-3421

### Behavior

After the skirt completes, the printer prints the tiny, and I mean almost microscopic; perimeters prior to moving on the larger outlines/perimeters.

Because these extrusions are so tiny, they often get pulled off by the retraction or lifting of the print head to move to the next tiny detail.

These become blobs on the nozzle and lead to failed prints.

Here is the first layer:

[image]

And zoomed in, the red circles show the tiny perimeters that are printing first - it is the 'dots' in the recurves of the 'S', etc:

[image]

And the resulting 'dots':

[image]

After many failed prints, with the first layer speed turned down to 12mm/s and then dialling the printer down to 50% speed during the perimeters and then back to 100% for the bulk infill. I was able to get a successful first layer to complete after 4 failures at this speed. The first layer takes around 4.5-5hrs doing it this way.

If the rest of the permeters printed first and then these tiny details were done after, they would have something to stick to without having to solely rely on bed adhesion.

#### Project File (.3MF) where problem occurs

[Switch Panel V2.zip](https://github.com/prusa3d/PrusaSlicer/files/5054825/Switch.Panel.V2.zip)

Thank-you and any assistance is greatly appreciated.

This is a new business venture for me and I really want my MK3S to be the start of a print farm.

Kadah, 2020-08-12:
A workaround for this I do for this problem is print the first layer or two with clear PETG and design the model so the features are subsurface by that same amount. This also has the added benefit making those MM features more durable.

KimmoHop, 2020-08-26:
If sliced with external perimeters first, is it better? There is no checkbox to do that only for the first layer :/

closed 2026-09-29 by prusaslicer-bot[bot]