prusa3d/PrusaSlicer #4625: Drainage drill offset wrong on scaled models

TheLogan, 2020-08-11:
### Version
2.2.0

### Behavior
I noticed this when I imported my model and discovered that it was 1/10th the scale I'd set in my 3D software (I belive this is an issue of my 3D software, not of Prusa).
So the fastest thing to do was select the model and set the scale factor to 1000%.

I then continued to hollow it out and add drainage holes, except that the drill ghost object was way off
As you can see it's hovering over the object
[image]

I then quickly changed the scale in my modelling software and re-imported, and this is the result, which is exactly what I'd expect 
[image]

The only difference between the two objects is that one was scaled before being imported the other scaled after.

#### Project File (.3MF) where problem occurs

[3DPrint.zip](https://github.com/prusa3d/PrusaSlicer/files/5056588/3DPrint.zip)

lukasmatena, 2020-08-12:
Reproduced, thanks. It will be fixed soon.

lukasmatena, 2020-08-24:
Hopefully fixed in `master` for 2.3 release - commit 20da7a817599354132df4a1946f65a0c8cdeede8.
Thanks again. Closing.

closed 2020-08-24 by lukasmatena