prusa3d/PrusaSlicer #4627: Feature Request: Portable installation option incl config, filament, profiles inside install directory.

mrstoned-gh, 2020-08-12:
I would like a portable install option.

I specify a directory for installation, and it installs the software, configs, profiles inside the install directory.
Similar to how Windows Commander by Ghisler stores it's configuratioin inside it's custom installation directory.

This would make for a portable installation on usb, or network install for configuration storage.

MooMinIL, 2024-01-02:
I know it's been 3 years since this issue was opened but I am also in a position where it could greatly benefit from the ability to install PrusaSlicer in portable mode on a usb thumbdrive/network drive or to run it in a computer where i do not have proper install permissions.

Really hope this is added in a future update, I assume it's not a large feature to add.

lukasmatena, 2024-01-02:
I know it is not exactly what the request is about, but you can already switch the configuration directory using the `--datadir` command line argument (you can create a shortcut that will pass the argument for you). And PrusaSlicer itself does not need any installation.

closed 2026-09-29 by prusaslicer-bot[bot]