prusa3d/PrusaSlicer #4631: Extrusion Width Percentge

buryboi, 2020-08-14:
### Version
2.2.0

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Artillery Sidewinder X1 / V4 + Marlin bugfix 2.0.x

### Behavior
Extrusion Width settings above 300% produce  an error popup. 

* _Steps needed to reproduce the problem_
Enable Expert Features, Click Print Settings, Click advanced and change any extrusion width to a number above 300%

  * _If this is a command-line slicing issue, include the options used_
N/A

* _Expected Results_
At a 0.2 mm layer height, a width of 400% should result in a layer width of 0.8mm as per the hover notes. "If expressed as a percentage it will be computed over layer height."  When 300% is used the layer width is correctly sliced at 0.6 mm.

[image]

* _Actual Results_
An error pops up stating "Excessive extrusion_width=1.6 mm to be printable with nozzle diameter 0.4 mm"

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Unable to slice and export until a number below 300% is input in the cell.

louwheesy, 2020-08-15:
It says why you can't do that. It's a rediculous extrusion width that cannot be reliably produced. It's not reccommended to print above 150%, although it lets you go to 300%. Defnitely not an issue.

n8bot, 2020-08-15:
If you pay closer attention to the issue described, he mentions how the tooltip describes the calculation acting over the layer height, when it seems to in fact act over the nozzle width.

buryboi, 2020-08-15:
I agree that 1.6 would be ridiculous. But 1.6 mm is 400% of nozzle width not layer height as the tool tip indicates.  Using 300% results in a sliced layer width of 0.6 mm which indicates that the actual slicing is done based on layer height as indicated by the tool tip. 

To summarize/reiterate: the tool tip indicates percentages will slice based on layer height. The error popping up is slicing based on nozzle width. When reviewing the sliced model, the width is in fact sliced based on layer height not nozzle width. Thus it is opinion that the error is incorrect.

rtyr, 2020-08-16:
Yes, it seems like a bug in sanity check (width is calculated over nozzle diameter instead of layer height) .

bubnikv, 2021-01-20:
Fixed with 6d7d482ca1388f7f95ecad7122248a4c431fa9b6.

closed 2021-01-20 by bubnikv