prusa3d/PrusaSlicer #4632: Add keep Changes dialog, when switching between Presets

Iqwertz, 2020-08-14:
I think it is extremly annoying that when I switch between presets all custom settings are reseted! This destroys the whole workflow! I want to be able to switch between presets and test which fits best for my print while still maintaining some custom parameter (like infill type)!

It would be great if the slicer had a cura like behaviour: When I change my preset a popup dialog should appear, which lists all my changes and I can deside if I want to keep or delete them. It should only look for changes that I made and didnt save to a preset! 

If you think this feature would be to confusing for beginner, it could be only enabled when expert modus is used or make it a tick box in the settings!

bubnikv, 2020-08-15:
@YuSanka is working on this feature right now.

Iqwertz, 2020-08-17:
Okay Thank you!

closed 2020-08-17 by Iqwertz