prusa3d/PrusaSlicer #4634: PrusaSlicer misinterpreting extrusion width percentages

louwheesy, 2020-08-15:
### Version
PrusaSlicer-2.2.0

### Operating system type + version
Windows 10 Home
### 3D printer brand / version + firmware version (if known)
Original i3 MK3s (not that it's important as this is a slicer issue)

### Behavior
When setting infill extrusion width for example, a value of 150% is used. The nozzle is set at 0.4mm, so this would mean that the width is 0.6mm. But when slicing it, the width appears much smaller. To verify, I set the width to a value of 0.6mm and it made the width much thicker in the sliced part. At the same time, if I set the width to 1000%, an error shows up stating that it cannot extrude a 4mm line, so it must be calculating the width properly, but not actually applying it.

[image]
[image]
[image]
[image]
[image]
[image]

neophyl, 2020-08-15:
Did you read the tooltip ?  Percentage setting is not done over nozzle width but over layer height. So if your layer height is 0.2 then 150% is 0.3 wide.

rtyr, 2020-08-15:
Like neophyl said, it is calculated over layer height. Closing.

[image]

closed 2020-08-15 by rtyr

louwheesy, 2020-08-15:
> Like neophyl said, it is calculated over layer height. Closing.
> 
> [image]

You didn't look at all of my pictures.
I'm printing at a layer height of 0.1mm.
So if I'm going off of a layer height multiple, 1000% of 0.1mm is 1mm. But when I set it to 1000%, it says that it cannot lay a width of 4mm. That means it's referencing my nozzle, which is the only thing that is 0.4mm.

[image]
[image]
[image]

rtyr, 2020-08-15:
Yes, value in the error message/sanity check is not correct, which was already reported here https://github.com/prusa3d/PrusaSlicer/issues/4631. Still, actual extrusion width is calculated over layer height.

louwheesy, 2020-08-15:
> Yes, value in the error message/sanity check is not correct, which was already reported here #4631. Still, actual extrusion width is calculated over layer height.

So there is no way to use nozzle diameter multipliers? It seems very cumbersome to have to create overlapping profiles for the same layer height with different nozzles, and then again for each material you use.

rtyr, 2020-08-15:
Not at the moment. Current implemenation is inherited from upstream Slic3r, it always worked that way. Related issue https://github.com/prusa3d/PrusaSlicer/issues/2829.