prusa3d/PrusaSlicer #4635: 1.41.2 heat bed and hotend at same time.

mchaos, 2020-08-16:
### Version
_Version of PrusaSlicer used goes here_
1.41.2
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
ubuntu 18.04
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Self Fabricated running current marlin bug fix.
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

Up until recently both my bed and hotend would heat at the same time. I prefer this. Starting yesterday it will heat the bed alone, then go to auto level and heat hotend while leveling.  

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

It's any gcode rendered. It shows that it has head the bed with blocking.. and heat nozzle unblocked...

Is there anyway to change it so that it generates the gcode so it either heats nozzle before bed so that they will both heat at the same time, and or change the heat command so that it will heat at the same time.

rtyr, 2020-08-16:
You can set it in the start g-code. 

Fox example like this: (hotend+bed at the same time):

```
M104 S[first_layer_temperature] ; set hotend temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for hotend temp
```

Closing.

closed 2020-08-16 by rtyr

mchaos, 2020-08-16:
I don't understand what changed.. I never had to do this.