prusa3d/PrusaSlicer #4636: Funky window drag move on Windows 10 v2004

Petttkous, 2020-08-16:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 64bit version 2004

### 3D printer brand / version + firmware version (if known)
Doesn't matter

### Behavior
Dragging the main window by the title bar shutters and is very delayed, laggy.

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
Happens with any models (even with a clean install and empty scene)

rtyr, 2020-08-18:
Can you please provide your system info? You can run _Help->System info_ from PrusaSlicer, then select "Copy to Clipboard" and paste it here.

[image]

AdamLeyshon, 2020-08-20:
Can confirm, trying to drag the window is very, very slow.

```
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 34,279MB
OpenGL installation
GL version:   4.6.0 NVIDIA 451.67
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2060/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
[... 323 lines cut ...]
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

```

3Delights, 2020-08-20:
Can also confirm, window lags behind movement and moves in an odd way.  Problem also present in the 2.3.0 alpha release!

```
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 68,629MB
OpenGL installation
GL version:   4.6.14736 Compatibility Profile Context 20.8.2 27.20.2001.16003
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon RX 5700 XT
GLSL version: 4.60
Installed extensions:
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_half_float_fetch
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_occlusion_query_event
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trace
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_sparse_texture_pool
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
[... 228 lines cut ...]
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
```

rtyr, 2020-08-25:
I was able to reproduce it on this system. Detailed system info attached.

[image]

[detail.zip](https://github.com/prusa3d/PrusaSlicer/files/5121546/detail.zip)

rtyr, 2020-08-25:
Are you guys aware of when this started to happen? Was it after some Windows update?

jmd73, 2020-08-26:
I got it after update to windows 10 2004 also, but I didn't see it initially. The reason I didn't get it from start, was that I had set my config to show the settings in separate windows instead of tabs. If I, in prefernces, set it to show the settings as the "old" way with tabs, it gets sluggish.

So if I set the preferences to show the settings in modal windows, it the sluggishness dissapears.

I am not at home right now, so I cant check the correct name for the preferences and send screenshots, but they are only available in the 2.3.0 branch. It is not available in 2.2.0 which I could test here at work (2.2.0 release is also slow since windows 10 2004.)

bubnikv, 2020-08-28:
It seems to be a breaking change after Windows 10 update in a way how WIN32 API presents the message queue during the window movement.

bool wxMSWEventLoopBase::Pending() const
{
    MSG msg;
    return ::PeekMessage(&msg, 0, 0, 0, PM_NOREMOVE) != 0;
} 

PeekMessage seems to return "Idle" even if the user drags the window around like a crazy monkey, even though the message queue should be full of mouse events. As our application redraws from OnIdle handler, this causes our application window to lag behind the  mouse cursor as if on a long soft rubber band.

We will likely modify our rendering behavior between the windows specific wxWidgets callback EVT_MOVE_START / EVT_MOVE_END.

I have reported this issue to wxwidgets list, but I think this time the Redmond guys are guilty.

bubnikv, 2020-08-28:
Hmm, the above does not seem to apply, I was wrong.

@jmd73 Thanks for the hint, we can reproduce that. When the parameter editor is floating, the parameter editor shows this rubbery effect when moved while the main application is moved swiftly.

bubnikv, 2020-08-31:
I am having the same effect with a 3 years old Slic3r PE on Windows build 2004, if I add couple more extruders, for which parameter pages are generated. For some reason, the Windows 2004 build does not like the sheer amount of Win32 controls at our parameter pages. We are considering to reduce them by custom rendering some of the static texts, images and the undo buttons.

I would complain to Microsoft, but I don't know where.

bubnikv, 2020-08-31:
I have reported to Microsoft using the Windows 10 built in Feedback Hub application. If you want Microsoft to fix it, please log into the Feedback Hub and upvote the following report:

Rubber band Window movement on Windows build 2004

After update to Windows build 2004, our Win32 application PrusaSlicer
https://github.com/prusa3d/PrusaSlicer
shows a weird effect, where the window lags behind the mouse cursor when dragging the window by a mouse.

This is the issue opened by our customers.
https://github.com/prusa3d/PrusaSlicer/issues/4636

I debugged the application and it really hangs inside the Win32 default window handler routine when dragging a window, our application does not hang inside our handler.

Our application creates a form with 1300 Win32 windows split into three tabs and multiple pages, see the attached screenshot. Most of the Win32 controls are hidden most of the time, however even these hidden windows seem to cause this rubber band window dragging effect. If we lower the number of Win32 controls 6x, the issue is fixed, however we did not have this issue before the build 2004 update.

jmd73, 2020-09-01:
@bubnikv I can't upvote it, since I can't find it on the feedback hub. The search function is, in my opinion, not very good.
There is another report there with the title "window movement is unbearably slow" that is regarding an app called "eartrumpet". I tried it and it has the same problem, but as far as I can tell, it doesn't have that many components on the forms.

formateins, 2020-09-08:
I'd like to confirm the rubber band effect on dragging the window. Also occurs on previous Win 10 Releases. Updated my Workstation last weekend to 1909 - same effect as on my 2004 main PC.
[sys-info.txt](https://github.com/prusa3d/PrusaSlicer/files/5190542/sys-info.txt)

bubnikv, 2020-10-07:
@jmd73 We have a solution prepared for PrusaSlicer 2.3 by @YuSanka, where the controls are populated on demand. It solves the rubber band window movement, but the switching of parameter pages is slower. @YuSanka will work on improving speed of switching the parameter pages.

God knows what the Microsoft guys screwed in the Windows build 2004.

bubnikv, 2020-10-09:
The issue will be fixed in PrusaSlicer 2.3.0-alpha1.
Closing.

closed 2020-10-09 by bubnikv

forkineye, 2021-11-09:
@bubnikv Do you know of a specific commit or PR in which this was fixed? Experiencing the same issue in another project with wxWidgets and am interested in how you mitigated it. I went through the commits but couldn't find anything. Thanks.

lukasmatena, 2021-11-09:
@forkineye I believe it was done between 398ff9053d2218bde0b365503b5d39ec18d40fa8 and b17c829c9ad3302e4960f16929fdd10698c4ad79. I'm not sure if you'll be able to take something from it. As far as I know, the main trick is to only create the controls on the current tab and destroy them when switching elsewhere, @YuSanka would know the details.

bubnikv, 2022-06-22:
@forkineye As far as I can remember, the issue was solved magically by replacing the wxWidgets notebook control with out own control. We did that mainly due to dark mode support as we were not able to skin the Win32 notebook properly.