prusa3d/PrusaSlicer #4643: Feature Request: "Silent Mode" Enhancement #2

sproggit, 2020-08-18:
### Version - 2.2.0_Linux_x64
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version - Linux Mint 19.2/64-Bit / Windows 10/64-Bit
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) Prusa i3/Mk 3S / 3.9.0-3421 / Stock Printer
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
Excessive noise, even when set to silent mode, when the printer is moving/relocating the print head back to the "start point" for a new print layer. Watching a printer operate in silent mode, the print head tends to move somewhat more slowly than in normal printing. However, when the printer "relocates" the print head, this movement is quite a bit faster - and noisier- than basic printing. 

My suggestion, very simply, is to ask if we could have a "super silent" print mode, which also slows down these non-printing movements, so that *everything* the printer does is completely silent... 

* _Steps needed to reproduce the problem_
Just run a print in silent mode, that includes the head relocating at the start of each layer... 

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
**Hoped-for** results - completely silent printing... 

* _Actual Results_
A little bit of extra noise when the print head moves across the object in non-printing, repositioning mode... 

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

This issue can be observed with any print, for example as per the file uploaded by my other, related issue, #4642. 

Thank you for your kind consideration!

rtyr, 2021-02-24:
Maximum feedrate and acceleration in silent mode is limited by the firmware.

As for the PrusaSlicer, you can create separate profile with slower speeds etc., or just separate printer profile with modified machine limits. There is no need for special feature, it is already possible.

neophyl, 2021-02-24:
I was going to suggest slowing his travel speed down in his slicer settings but I like your suggestion better :)

closed 2026-09-29 by prusaslicer-bot[bot]