prusa3d/PrusaSlicer #4644: Post Process scripts fails when run from PrusaSlicer

Jacotheron, 2020-08-18:
### Version
PrusaSlicer version 2.2.0

### Operating system type + version
Windows 10 build 1909

### Behavior
I created a post process script (both in perl and python, after I got the issue), which simply replace T- commands for use on my single extruder machines. Both the Perl and Python scripts work when I call them in cmd, powershell and git-bash, with the form: "<script-path-and-name> <gcode-path-and-name>" (no need for me to even prepend for example "py").

However when I run it from PrusaSlicer, the python get error: Failed starting the script <path>\duet-toolchange-strip.py <path>\dual-test8.gcode, Win32 error: 193
The Perl I get error: Post processing script <path>\duet-toolchange-strip.pl on file <path>\dual-test8.gcode failed. Error code: 25

I have attempted to fix both of the issues, spending many hours, but is unable to.

My Python script is as follows:
```
#! C:\Python38\python.exe
# duet_toolchange-strip.py
import fileinput
import re
import sys

extruder_0_pattern = re.compile('^T0')
empty_line = re.compile('^$')
extruders_other_pattern = re.compile('^T[0-9]+')
extruders_other_pattern_catch = re.compile('^T([0-9]+)')
print_have_started = 0
for line in fileinput.input(inplace=True):#, files=(sys.argv[1])
    if extruders_other_pattern.match(line):
        if (extruder_0_pattern.match(line) and print_have_started == 0):
            print(line.rstrip("\n"))
        else:
            print_have_started = 1;
            m = extruders_other_pattern_catch.match(line)
            extruder_nr = m.group(1);
            print('M291 R"Tool Change" P"Tool #%s " S1 T30\nM600' % extruder_nr)
    else:
        if not empty_line.match(line):
            print(line.rstrip("\n"))
```

and my Perl is as follows:
```
#!C:\Strawberry\perl\bin\perl.exe -i -p
#
# Remove all toolchange commands (and T# arguments) from gcode.

# use strict;
# use warnings;

 my $print_have_started = 0;
# read stdin and any/all files passed as parameters one line at a time
while (<>) {
  if (not /^T[0-9]+/) { #this is the standard option for each line, that is not a tool change
    s/\s*(T[0-9]+)//; #this will strip out tool change mid of lines - I am not going to cover this
    print; #output the line
  } else {
    if(/^T0/ && $print_have_started == 0){
      print;
    }else{
      $print_have_started = 1; #only once we get a T1, should we start the stripping
      s/^T([0-9]+)/M291 R"Tool Change" P"Tool #$1" S1 T30\nM600/; #match a toolchange and insert message and M600
    print; #output the line
    }
  }
}
```

I only have a 64bit python installed, also perl is 64bit.

While creating this, I used Slic3r scripts as a start, but even those scripts don't work.

I am thinking the issue might be exactly how the child process is created (in current master src/libslic3r/gcode/postprocessor.cpp on line 164). How can I simulate this in any of my terminals/shells to ensure it works?

foreachthing, 2020-08-20:
You should be able to use your python script directly in a console.
Start the script and pass your cgode file as parameter.

Example:
`python yourscript.py your.gcode`

Jacotheron, 2020-08-20:
I had tested my code exactly as your example shows, and it worked (I even got Windows to allow executing the script and calling Python automatically without the 'python' at the beginning).

The issue was that I did not realize on Windows the path to the python executable should have spaces prepended with "!" instead of simply enclosing the whole path in quotes (as is done usually) - this is in the Slic3r manual, in the link above. I have now tested through PrusaSlicer and it simply worked.

Thank you for the assistance.

closed 2020-08-20 by Jacotheron

touradnik9, 2021-01-07:
Can you elaborate on how this problem was fixed? I am getting the same error. Where do I change things and what do I add? Thanks!

Jacotheron, 2021-01-07:
In short I used full path to the python executable (in my case it is a v3.8.X, which is located at "C:\Python38\python.exe") followed by a space and then the full path to the post processor script (note that if this path contains any spaces, you should enclose it with double quotes). Therefore my full line for the post-processor script is:

`C:\Python38\python.exe <path-to-script>\duet-toolchange-strip.py;`

This worked for me.

Gallion, 2025-03-01:
I've tried the following : 
C:\Program Files\Python310\python.exe D:\3D pranting\Print this\Tests\LineTestPost.py; -> The configured post-processing script does not exist: C:\Program
"C:\Program Files\Python310\python.exe D:\3D pranting\Print this\Tests\LineTestPost.py"; -> The configured post-processing script does not exist: C:\Program Files\Python310\python.exe D:\3D pranting\Print this\Tests\LineTestPost.py
"C:\Program! Files\Python310\python.exe D:\3D! pranting\Print! this\Tests\LineTestPost.py"; -> The configured post-processing script does not exist: C:\Program! Files\Python310\python.exe D:\3D! pranting\Print! this\Tests\LineTestPost.py
C:\Program! Files\Python310\python.exe D:\3D! pranting\Print! this\Tests\LineTestPost.py; -> The configured post-processing script does not exist: C:\Program!
"D:\3D pranting\Print this\Tests\LineTestPost.py"; -> Failed starting the script "D:\3D pranting\Print this\Tests\LineTestPost.py" C:\Users\MiniPC\AppData\Local\Temp\.PrusaSlicer.upload.a2a5-f36b-fde7-01e3, Win32 error: 193

I'm at my wit's end. How do I actually run this??

Gallion, 2025-03-01:
For anyone wondering, the proper format when the paths contain spaces is 
"Path\To\Python.exe" "Path\to\script";