prusa3d/PrusaSlicer #4646: problème de G-Code

simonvolland, 2020-08-18:
### Version
 2.2.0+

### Operating system type + version
Mac OS CATALIANA 10.15.5

### 3D printer brand / version + firmware version (if known)
 CREALITY CR10 V2 Marlin 

### Behavior

G-code export to /var/folders/cj/gymmlkqd0lj1jh41c9tmfnf00000gn/T/.32334.gcode failed due to invalid custom G-code sections:
end_gcode
start_gcode
Please inspect the file /var/folders/cj/gymmlkqd0lj1jh41c9tmfnf00000gn/T/.32334.gcode.tmp for error messages enclosed between
!!!!! Failed to process the custom G-code template ...
and
!!!!! End of an error report for the custom G-code 

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs

When i want to slice my STL

foreachthing, 2020-08-19:
Please attach (zip, 7zip) this file: /var/folders/cj/gymmlkqd0lj1jh41c9tmfnf00000gn/T/.32334.gcode.tmp

rtyr, 2021-03-22:
No response. Closing.

closed 2021-03-22 by rtyr