prusa3d/PrusaSlicer #4647: PrusaSlicer master segfault

pmjdebruijn, 2020-08-19:
### Version
2.3.0~alpha0+315~g5052149b8 (current master)

### Operating system type + version
xubuntu 20.04

### Behavior
When starting PrusaSlicer master, it segfaults.

What's remarkable is when I move .PrusaSlicer-alpha out of the way, prusa-slicer will start once, but will segfault again once .PrusaSlicer-alpha is present.

### gdb output
```
$ gdb prusa-slicer
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from prusa-slicer...
(No debugging symbols found in prusa-slicer)
(gdb) r
Starting program: /usr/bin/prusa-slicer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffefe8e700 (LWP 52406)]
[New Thread 0x7fffef68d700 (LWP 52407)]
[Detaching after vfork from child process 52408]
[Detaching after vfork from child process 52410]
[Detaching after vfork from child process 52412]
[Detaching after vfork from child process 52414]
[Detaching after vfork from child process 52416]
[Detaching after vfork from child process 52418]
[Detaching after vfork from child process 52420]
[Detaching after vfork from child process 52422]
[New Thread 0x7fffecf96700 (LWP 52424)]
[New Thread 0x7fffdb96b700 (LWP 52425)]
[New Thread 0x7fffdb16a700 (LWP 52426)]
[New Thread 0x7fffda969700 (LWP 52427)]
[New Thread 0x7fffda168700 (LWP 52428)]
[New Thread 0x7fffd9967700 (LWP 52429)]
[New Thread 0x7fffd9166700 (LWP 52430)]
[New Thread 0x7fffd8965700 (LWP 52431)]
[New Thread 0x7fffc3fff700 (LWP 52432)]
[New Thread 0x7fffc37fe700 (LWP 52433)]
[New Thread 0x7fffc2ffd700 (LWP 52434)]
[New Thread 0x7fffc27fc700 (LWP 52435)]
[New Thread 0x7fffc1ffb700 (LWP 52436)]
[New Thread 0x7fffc17fa700 (LWP 52437)]
[New Thread 0x7fffc0ff9700 (LWP 52438)]
[New Thread 0x7fff9ffff700 (LWP 52439)]
[New Thread 0x7fff9f7fe700 (LWP 52440)]
[New Thread 0x7fff9effd700 (LWP 52441)]
[New Thread 0x7fff9e7fc700 (LWP 52442)]
[New Thread 0x7fff9dffb700 (LWP 52443)]
[New Thread 0x7fff9d7fa700 (LWP 52444)]
[New Thread 0x7fff9cff9700 (LWP 52445)]
[New Thread 0x7fff7ffff700 (LWP 52446)]
[New Thread 0x7fff7f7fe700 (LWP 52447)]
[New Thread 0x7fff7effd700 (LWP 52448)]
[... 142 lines cut ...]
#12 0x00007ffff7283bd1 in wxEvtHandler::ProcessEvent(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#13 0x00007ffff70f960c in wxAppConsoleBase::ProcessIdle() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#14 0x00007ffff6d40f18 in wxAppBase::ProcessIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#15 0x00007ffff6c6b6a5 in wxApp::DoIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff6c6b7a7 in  () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff4a9de8e in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff4a9e240 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007ffff4a9e533 in g_main_loop_run () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007ffff50d637d in gtk_main () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#21 0x00007ffff6c89575 in wxGUIEventLoop::DoRun() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#22 0x00007ffff7137d41 in wxEventLoopBase::Run() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#23 0x00007ffff70ff3da in wxAppConsoleBase::MainLoop() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#24 0x00007ffff718b73d in wxEntry(int&, wchar_t**) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#25 0x0000555555733d3d in  ()
#26 0x0000555555714df9 in  ()
#27 0x00007ffff63320b3 in __libc_start_main (main=0x555555714cb0, argc=1, argv=0x7fffffffdff8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdfe8) at ../csu/libc-start.c:308
#28 0x000055555572d92e in  ()
(gdb) q
A debugging session is active.

	Inferior 1 [process 52402] will be killed.

Quit anyway? (y or n) y

```

pmjdebruijn, 2020-08-23:
And now with proper debug symbols:

```
Thread 1 "prusa-slicer" received signal SIGSEGV, Segmentation fault.
0x0000555556302557 in ImFont::CalcTextSizeA (this=0x0, size=0, max_width=max_width@entry=3.40282347e+38, wrap_width=-1, text_begin=text_begin@entry=0x55555638f592 "A", text_end=0x55555638f593 "", remaining=remaining@entry=0x0)
    at ./src/imgui/imgui_draw.cpp:2838
2838	./src/imgui/imgui_draw.cpp: No such file or directory.
(gdb) bt
#0  0x0000555556302557 in ImFont::CalcTextSizeA(float, float, float, char const*, char const*, char const**) const
    (this=0x0, size=0, max_width=max_width@entry=3.40282347e+38, wrap_width=-1, text_begin=text_begin@entry=0x55555638f592 "A", text_end=0x55555638f593 "", remaining=remaining@entry=0x0) at ./src/imgui/imgui_draw.cpp:2838
#1  0x00005555562d7c87 in ImGui::CalcTextSize(char const*, char const*, bool, float)
    (text=text@entry=0x55555638f592 "A", text_end=text_end@entry=0x0, hide_text_after_double_hash=hide_text_after_double_hash@entry=false, wrap_width=wrap_width@entry=-1) at ./src/imgui/imgui.cpp:4218
#2  0x0000555555e7407f in Slic3r::GUI::NotificationManager::PopNotification::init() (this=0x5555594ba800) at ./src/slic3r/GUI/NotificationManager.cpp:186
#3  0x0000555555e749a9 in Slic3r::GUI::NotificationManager::PopNotification::PopNotification(Slic3r::GUI::NotificationManager::NotificationData const&, int, wxEvtHandler*)
    (this=0x5555594ba800, n=..., id=<optimized out>, evt_handler=0x5555574ce050) at ./src/slic3r/GUI/NotificationManager.cpp:52
#4  0x0000555555e78235 in Slic3r::GUI::NotificationManager::push_notification_data(Slic3r::GUI::NotificationManager::NotificationData const&, Slic3r::GUI::GLCanvas3D&, int) (this=0x5555567f6800, notification_data=..., canvas=..., timestamp=0)
    at ./src/slic3r/GUI/NotificationManager.cpp:791
#5  0x0000555555ea1515 in Slic3r::PresetUpdater::config_update(Slic3r::Semver const&, bool) const (this=this@entry=0x55555685a870, old_slic3r_version=..., no_notification=no_notification@entry=false)
    at ./src/slic3r/Utils/PresetUpdater.cpp:829
#6  0x0000555555c624bb in Slic3r::GUI::GUI_App::check_updates(bool) (this=0x55555679f890, verbose=<optimized out>) at ./src/libslic3r/Semver.hpp:72
#7  0x0000555555c626a2 in Slic3r::GUI::GUI_App::<lambda(wxIdleEvent&)>::operator() (__closure=<optimized out>, event=...) at ./src/slic3r/GUI/GUI_App.cpp:472
#8  wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::<lambda(wxIdleEvent&)> >::operator()(wxEvtHandler *, wxEvent &) (this=0x5555575b5ee0, event=...) at /usr/include/wx-3.0/wx/event.h:516
#9  0x00007ffff7282641 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#10 0x00007ffff72829da in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#11 0x00007ffff7282a74 in wxEvtHandler::TryHereOnly(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#12 0x00007ffff7282b2b in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#13 0x00007ffff7282bd1 in wxEvtHandler::ProcessEvent(wxEvent&) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#14 0x00007ffff70f860c in wxAppConsoleBase::ProcessIdle() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#15 0x00007ffff6d3ff18 in wxAppBase::ProcessIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#16 0x00007ffff6c6a6a5 in wxApp::DoIdle() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#17 0x00007ffff6c6a7a7 in  () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#18 0x00007ffff4a9ce8e in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#19 0x00007ffff4a9d240 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#20 0x00007ffff4a9d533 in g_main_loop_run () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff50d537d in gtk_main () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#22 0x00007ffff6c88575 in wxGUIEventLoop::DoRun() () at /lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
#23 0x00007ffff7136d41 in wxEventLoopBase::Run() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#24 0x00007ffff70fe3da in wxAppConsoleBase::MainLoop() () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#25 0x00007ffff718a73d in wxEntry(int&, wchar_t**) () at /lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0
#26 0x0000555555733d7d in Slic3r::CLI::run(int, char**) (this=<optimized out>, argc=<optimized out>, argv=<optimized out>) at ./src/PrusaSlicer.cpp:562
#27 0x0000555555714e39 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/include/c++/9/bits/stl_vector.h:94
```

pmjdebruijn, 2020-08-23:
Presumably b3f8ae5ca75b52ef74e8da2be2392ea8339dcab2 is related?

pmjdebruijn, 2020-08-23:
@kocikdav:

Also interesting, when I set preset_update = 0 in ~/.PrusaSlicer-alpha/PrusaSlicer.ini, PrusaSlicer no longer segfaults, so the issue is triggered by a new set of ini's being made available. If the local ini's are fully up to date, this will not trigger apparently.

lukasmatena, 2020-08-25:
Thanks for the backtrace, the additional info has helped. I think I see the problem. I will ping @kocikdav who will know what can and cannot be done.

kocikdav, 2020-08-26:
@pmjdebruijn Thanks for backtrace. Unfortunately I couldn't reproduce the issue but I think its fixed now. Soon in master.

kocikdav, 2020-09-02:
In master branch now. Im closing this issue. Please reopen if the problem would still occur.

closed 2020-09-02 by kocikdav