prusa3d/PrusaSlicer #4648: Info-box: Volume wrong after scaling

foreachthing, 2020-08-20:
### Version
Commit #739cd2a

### Operating system type + version
Win 10 x64 1809

### 3D printer brand / version + firmware version (if known)
Ultimaker 2

### Behavior
* Import model (I made a 10x10x10 mm cube)
  - check volume in Info box
* scale model (to any scale you like, i.e. 112%)
  - check volume again - it's wrong!

The volume calculation is kept from the original imported file:
[image]

Also, after scaling an object, the size is not refreshed. One needs to deselect this object first, then after selecting it again, it will be updated.

Also2: it should not be called "scale factor" as it is a percentage, it should read "scale" only. IMHO.

rtyr, 2020-08-20:
Looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4457. Closing.

closed 2020-08-20 by rtyr

foreachthing, 2020-08-20:
Sorry, didn't see that one!

rtyr, 2020-08-20:
No problem :)