prusa3d/PrusaSlicer #4652: [Feature Request] Infill "perimeters"

albert2004, 2020-08-21:
### Version
all

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Currently, infill is generated with 1 line/perimeter. This works fine in almost all cases. But, if I (or someone else) want to create case with, for example, honeycomb top/bottom, having a possibility to have it in slicer would be great.
It is possible to model honeycomb, but:
1. It takes time (don't argue).
2. Printing results in a lot of perimeters around honeycomb and retract/un-retract.
When, on solid object I set 0 top and bottom layers, I can have honeycomb pattern, but with only one perimeter (for example 0.45mm width). Strength can be not enough for object. In this case it should be possible to set more "perimeters" for infill - to have it wider. It can be done by changing infill width but result depends on material and nozzle and cannot be extended infinitely. It is also faster.

Why I believe having this option is better than modeling "correct" object?
Because infill extruding is optimized for constant extrusion - honeycomb can be archived with minimum number of retractions an with almost no artifacts created when nozzle stops/retracts/moves.

Example honeycomb with current possibilities - nozzle 0.4, width 0.9mm:
[image]

Jackowski3k, 2022-06-30:
I agree (and also wonder why no one has done anything, or at least asked about it in last few yers). 

Having a way to change the infill thickness would let for much more interesting designs:
- strong cases with nice airflow due to bigger holes
- prettier decorative parts
- sturdier design than single-wall infilled print at same amount of filament used

I have made a simple comparison pics here to show the general idea. 
Both parts are around 60g (according to prusa slicer), one has 3 walls for infill, other standard honeycomb pattern. 
[image]
[image]

sglux, 2022-07-04:
I'm thinking about a comparative print to know/show by experiment how far stiffness can be increased.
But it's not trivial to get a proper slicing for the multi-perimeter-infill with the slicer I have ;-)

mjack3k, 2026-09-18:
NGL it would be nice to have this feature