prusa3d/PrusaSlicer #4653: Graphics corruption

ITGuruWoo, 2020-08-21:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 2004 19041.450
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
WorkingSet: 287MB
PrivateBytes: 133MB
Pagefile(peak): 133MB(692MB)
RAM size reserved for the Undo / Redo stack: 1,074MB
RAM size occupied by the Undo / Redo stack: 0MB
OpenGL installation
GL version: 4.5.0 - Build 25.20.100.6519
Vendor: Intel
Renderer: Intel(R) UHD Graphics 630
GLSL 

Prusa MK3 S

### Behavior
Display is corrupted. Looks like graphics driver issue maybe
[image]

enricoturri1966, 2020-08-21:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4377 ?

JWIIwork1, 2020-09-10:
Yes, I believe this is a duplicate of the issue I am having.  The issue documented in #4377.  Issue is still open and no new actions to try and resolve.

lightingcze, 2023-03-27:
I have same issue :-( Im using all actual drivers, nothing help to me.

lukasmatena, 2023-03-27:
> I have same issue :-(

Which PrusaSlicer version are you running? Do you have this problem even with 2.6.0-alpha5?

lukasmatena, 2024-02-18:
I guess that this was solved by some of the OpenGL related changes in recent releases. Closing.

closed 2024-02-18 by lukasmatena