prusa3d/PrusaSlicer #4663: Prusa Slicer repeats full last layer after manual color change

fscheps, 2020-08-23:
### Version
2.2.0+Win64

### Operating system type + version
Microsoft Windows [Version 10.0.19041.450]

### 3D printer brand / version + firmware version (if known)
Prusa MK3S Firmware 3.9

### Behavior
I needed to print an even base of a specific color and then change color for when the letters start printing. For this I configured a manual color change on the Prusa slicer:

On this quick screen capture you can see how the color change is configured on Prusa Slicer: 
https://photos.app.goo.gl/qcMtfdNa4DfFCdTx7

On this other video you can see what the printer is actually printing after I changed color. Instead of starting with the letter, it continued to do another layer of the little squares. 
https://photos.app.goo.gl/VhuDkHv2En3YcTXTA

What am I expected for the print to come out?
Print 4 x 1 mm squares that will serve as the background for letters. Then after this layer of light green color in my print). Change filament color for gray and start printing the letters. Instead, what is happening is that the printer prints another layer of the squares in the gray color and only then starts printing the letters. Deceiving the purpose of having the green color in the background. 

Here a picture of the actual outcome: 
https://photos.app.goo.gl/dBrz49ioJsuzf8uX6

I am honestly not sure what is going on, I think I am doing everything right, but I am quite new to this 3D Printing world after I assembled my MK3S from the kit. 

I am sending the G Code from Prusa Slicer directly to OctoPrint 1.4.2 Python 2.7.16 OctoPi 0.17.0

Thanks a lot for your help to understand if this is potential bug or an issue of another sort.

Cheers

Fernando

[3MF_and_GCode.zip](https://github.com/prusa3d/PrusaSlicer/files/5115103/3MF_and_GCode.zip)

[image]

rtyr, 2020-08-24:
I checked your 3MF/gcode and I don't see any issue. I can see, that you inserted color change at layer 5 (1.0), it means that this layer will be already printed with the new color (M600 command is inserted at the beginning of that layer NOT at end). You can see real representation on your first video (dark red color on the model), which matches with your actual print. Color change should be added to layer 6 (1.2) in this case.

This is your 3MF:
[image]

This is color change at layer 1.2 (which is the layer where the letters begins):
[image]

fscheps, 2020-08-24:
Thanks a lot, your tip indicating "new color has to be inserted at the beginning and not at the end" is exactly what makes the whole difference. Thanks a lot! Will keep this in mind and will close this as an issue because its clearly not. Thank you.

closed 2020-08-24 by fscheps