prusa3d/PrusaSlicer #4665: Bug: Can't have square bracket inside comments in start_code

phazei, 2020-08-25:
### Version
V 2.2.0 + Win64

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Custom, marlin

### Behavior
If I put in any Start G-Code that has a "[ ]" that's commented it gives this error:

!!!!! Failed to process the custom G-code template start_gcode
Parsing error at line 18. Expecting tag literal-char
; Ensure extruder is not reset by other code or it will be 3mm short [see next line also]
                                                                          ^
!!!!! End of an error report for the custom G-code template start_gcode

lukasmatena, 2020-08-25:
This is not a bug. Square brackets are legacy Slic3r syntax for placeholders, so the parser is trying to substitute the bracket for a valid expression. And of course fails.

Just do this:

phazei, 2020-08-26:
Since it's in a comment, what would a comment need a placeholder for?  Maybe it's a bug in Slic3r then that is trying to parse placeholders in comments.  I would still contend it's a bug since nothing in the comments would ever be run and it wouldn't break anything to not parse brackets in comments.

n8bot, 2020-08-26:
There are many reasons to use a placeholder in a comment. E.g., someone may want to change the way the build time is formatted to pass it to their firmware in a way the firmware understands.

lukasmatena, 2020-08-27:
I agree with @n8bot. There are valid reasons to do expression substitutions in comments and I am quite certain that there are people who use it to pass something to their post-processing scripts, their custom/modified firmwares or just to make their gcodes more readable. If we do not parse comments, we will break it for them. On the other hand, I see no benefit in doing that (i.e., no actual issue that it would solve).

lukasmatena, 2020-09-01:
I'm closing this issue, I do not consider it to be a bug.
It can be reopened if someone convinces us otherwise.

closed 2020-09-01 by lukasmatena

chrishamm, 2021-09-05:
@lukasmatena RepRapFirmware supports [conditional G-code](https://duet3d.dozuki.com/Wiki/GCode_Meta_Commands) which could be mixed with PrusaSlicer expressions. I'd like to use something like this in my start G-code:

```
if move.extruders\[0\].filament != "[filament_type]"
  abort "Wrong filament loaded, this print was sliced for [filament_type]!"
```

For now I'll just move this section to a separate macro and invoke it with a parameter instead, I think that should do the trick as well. But perhaps this is a good example why it would make sense to be able to escape brackets.

lukasmatena, 2021-09-06:
@chrishamm

bubnikv, 2021-09-06:
@lukasmatena Great idea. I asked our content team to update https://help.prusa3d.com/en/article/macros_1775 accordingly.