prusa3d/PrusaSlicer #4666: Shift+Drag selection of support points in manual mode doesn't work

soswow, 2020-08-25:
### Version
2.3.0-alpha

### Operating system type + version
MacOS

### Behavior
* Shift+Drag selection of support points in manual mode doesn't always (or more like rearly) works. Not sure what's the algorithm, but it seems to be selecting random points.

https://youtu.be/xv38HTBOeuA

lukasmatena, 2020-08-25:
Can you share that 3MF and provide exact list of instructions to reproduce it? I tried on several models but never saw it.
But you're not the first person mentioning that.

soswow, 2020-08-25:
[simple-cube-project.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5122299/simple-cube-project.3mf.zip)

New video with voice over of how to replicate with a simple cube. https://youtu.be/4IuO5sX_Jzg (might still be processing)

lukasmatena, 2020-08-25:
Great, thank you, I can now reproduce it reliably, so I should be able to find and fix it.

soswow, 2020-08-25:
Great! Thank you!
The reason I want that very badly is to be able to remove supports from inside of a model (after hollowing).
Btw, @lukasmatena what is the best place to ask some technical questions (that only developer can answer)? For examples, in my case - I don't want all that pesky supports filling in internals after hollowing. Example:

[image]

As I understand the way you place support points determined not only by islands, but also by the weight and peeling/suction forces  etc. Question is - if I just remove these points (that are inside) manually - do I break that balance that was used in automatic point placement (and hence me reducing required support and hence increasing chances of a failure)

lukasmatena, 2020-08-25:
> what is the best place to ask some technical questions (that only developer can answer)?

lukasmatena, 2020-08-25:
@soswow 
> what is the best place to ask some technical questions (that only developer can answer)?

Here on github. But do not mix various issues together like you did now. Anyway:

> The reason I want that very badly is to be able to remove supports from inside of a model (after hollowing).

The huge amount of supports in the hollowed cavities is a bug. This is currently being fixed and optimized by one of the developers.

> if I just remove these points (that are inside) manually - do I break that balance that was used in automatic point placement

Technically you do, but the algorithm is probably not as smart as you. And like I said, at least in case of hollowed cavities it is even buggy, so remove whatever you want. It is best to think about the autoplacement as a default for manual editing, not as something that will "just print" (although it usually does).

soswow, 2020-08-25:
> But do not mix various issues together like you did now

Sorry about that. Was just giving an example of a Q. I have couple of more issues that I will post separately now.

> This is currently being fixed and optimized by one of the developers.

Is there an issue for that I can track?

lukasmatena, 2021-09-22:
Should be fixed with 284721e for PrusaSlicer 2.4.0-alpha1. Please, retest after it is released.
It turned out to be much simpler and more embarrassing than I thought it would be. The bug was in the function checking whether the point is obscured by another part of the mesh, the direction to the camera was provided in incorrect coordinate system. That's why it affected the whole face of the cube at once.
Closing.

closed 2021-09-22 by lukasmatena