prusa3d/PrusaSlicer #4667: Feature request: edit first color in multicolor (single extruder) print

CrustaShrimp, 2020-08-25:
### Version
PrusaSlicer 2.2.0+win64

### Operating system type + version
Windows 10 (64 bits)

### 3D printer brand / version + firmware version (if known)
Irrelevant

### Behavior
(Feature request)
I really like the current possibility of printing multicolor objects with a single extruder by using layerheight as a switch.
The preview is really good, where you can use right-click to edit the color of that layer group.
However, it does not work on the lowest layer!
This makes sense because the first layer will use the filament color in the filament settings, but it would be way more convenient and easy to use if the "bottom" color can also be edited by rightmouse option. This would then include the change in the filament settings on the filament tab.

brettvitaz, 2020-08-26:
Click on the color swatch for the filament in the right side. It will present you with the color picker that works for the first layer.

CrustaShrimp, 2020-08-26:
Oh that's really neat!
Although it might still be nice to have the same consistency in the different layer groups, especially with the idea of hiding the right panel in later releases, as was shown in [Prusa Live #3](https://youtu.be/NAW2JR-D_QU?t=2861).
No rush on this tho, and thanks again for showing the way of changing the filament color!

rtyr, 2021-02-24:
The current behavior seems correct to me. Closing.

closed 2021-02-24 by rtyr