prusa3d/PrusaSlicer #4668: Feature request: Toggle off Jerk limits so we can use Linear advance without editing G code.

PigeonFX, 2020-08-25:
Version
2.2.0+win64

Operating system type + version
All

Behavior
This is a feature request.

Description
The option to disable Jerk limits in 'Machine Limits' and having M205 automatically added to the Gcode, so we can use Linear advance without editing G code.

rtyr, 2020-08-25:
Why would machine limits prevent you from using linear advance?

PigeonFX, 2020-08-25:
> Why would machine limits prevent you from using linear advance? Are your prints slower (lower acceleration) with jerk limits? If so, your extruder jerk is set incorrectly. Set it to the same value as in your firmware.
> 
> Anyway, if you still don't want machine limits, you can easily disable it by switching g-code flavor to "Reprap/Sprinter".

Sorry Long day meant junction deviation, not Linear advance, due to PrusaSlicer adding Jerk values into the Gcode overwriting the Linear advance settings, but guess the switching g-code flavor would fix that too by taking out the M20x settings?

rtyr, 2020-08-25:
>  due to PrusaSlicer adding Jerk values into the Gcode overwriting the Linear advance settings

I am not sure what do you mean by that. I believe that M205 commands (XYZ) are ignored by printer firmware if you are using junction deviation, except extruder jerk, which should be set according to your firmware.

If you switch gcode flavor to Reprap/Sprinter, machine limits will be completely disabled and M20X commands will not be added.

PigeonFX, 2020-08-25:
> > due to PrusaSlicer adding Jerk values into the Gcode overwriting the Linear advance settings
> 
> I am not sure what do you mean by that. I believe that M205 commands (XYZ) are ignored by printer firmware if you are using junction deviation.

I have been told the reverse, so not sure where to go for clarity on this, 

> 
> If you switch gcode flavor to Reprap/Sprinter, machine limits will be completely disabled and M20X commands will not be added.

Yes, this sounds like a workaround that will work, but its not so much a "Feature" or intuitive. A tick box in the machine limits would be better IMHO.

rtyr, 2020-08-25:
You can test it for example via Pronterface.

There is already feature request (issue) for it https://github.com/prusa3d/PrusaSlicer/issues/4107.

rtyr, 2020-08-25:
So I believe we can close this.

closed 2020-08-25 by PigeonFX