prusa3d/PrusaSlicer #4669: Feature request: Failed object ignore

androandra, 2020-08-26:
When printing multiple parts in the same print, failure due to poor adhesion, warping or other stuff is often only related to a single part. However you still have to stop the whole print nonetheless, destroying all unifinished parts in the print. 

I'd like a feature in the printer UI so that you can choose a given part on the bed to ignore from that point onwards so the remaining parts can finish regardless. With this feature you wouldn't have to stop the print completely if you happen to catch the printer failing mid-print on just one part.

Of course the link between a snippet of G-code and its associated part would have to be encoded into the .gcode file in the slicing stage, so the printer can know which G-code to ignore. 

PrusaSlicer could output a temporary image file with a top-view, arbitrarily numbering the parts on the bed. This allows you to go back to your computer upon print failure, and ensure that the part number selected for ignoring is indeed the right one. This image file would be overwritten at every new slicing to save space. 

_Is this a new feature request?_
Yes

neophyl, 2020-08-26:
If you want this added to the **printer** UI/firmware then this really needs to be requested in the repository for that printers UI/firmware.  There are separate ones for the MK3 and the Mini(Buddy).  Located https://github.com/prusa3d/Prusa-Firmware-Buddy/issues and https://github.com/prusa3d/Prusa-Firmware/issues

Prusa Slicer can do this already at the  moment but only in conjunction with something like Octoprint and you have to enable verbose gcode output iirc.  Octoprint can then cancel individual items from the printing process as it handles sending the gcode to the printer and it can filter what it sends.

androandra, 2020-08-26:
Right, thanks! Will close this one and open up another in the right forum.

closed 2020-08-26 by androandra