prusa3d/PrusaSlicer #4675: Modifier Mesh Priorities

lanceofwhichwedream, 2020-08-28:
### Version
2.2.0+linux+x64
### Operating system type + version
Ubuntu 20.04.1
### 3D printer brand / version + firmware version (if known)
MK3S with MMU2S

### Behavior
(New Feature Request)

Allow modifier meshes to have a priority system where one mesh's effects can override the effects of another mesh.

Example: I was using the change extruder attribute on a mesh to "paint" a model. The "painted" part was a half oval, while I could have created my own mesh object, I also could have only painted the half oval had a feature adding me to state that one mesh's extruder value overrode or had a higher priority than another mesh's extruder for a specific area.

bubnikv, 2020-08-28:
There is such priority in the order they appear in the object list. You can
reorder them by drag & drop.

lanceofwhichwedream, 2020-08-29:
Interesting i found that i had issues with it, ill post the stl and project files here tomorrow. Maybe I'm doing something wrong.

lanceofwhichwedream, 2020-08-30:
So I played around a little bit and I found that the override works, but only if it's a extruder that isn't the original extruder used for the main body which doesn't seem like ideal behavior. I've attached a 3mf file for an example.
[PikachuThick_tail.zip](https://github.com/prusa3d/PrusaSlicer/files/5147606/PikachuThick_tail.zip)

puckpuck, 2020-09-30:
I have the exact same issue.  attached is another example.  In this case the `blue-H` modifier should appear on the sliced view but you don't see it. 
[MTL-logo.zip](https://github.com/prusa3d/PrusaSlicer/files/5303170/MTL-logo.zip)

dojohnso, 2023-07-20:
found this looking for a solution to my issue. I want to add a custom shape modifier from a loaded stl. i want to use that to change the extruder in that area. but the part of the model I'm adding it to is painted different from the default extruder, and the pain is override the model, but I want to be able to have the model override the paint .

base color and modifier (set to 2nd color)
[image]

painted color of base in 2nd color
[image]

sliced
[image]

closed 2026-09-29 by prusaslicer-bot[bot]