prusa3d/PrusaSlicer #4677: Sometime when previously "exported with supports" model is hollowed out - half of faces got inverted [SLA]

soswow, 2020-08-28:
### Version
2.3.0-alpha0

### Operating system type + version
MacOS

### Behavior
Since there is a known bug (that I was promised is currently being fixed) with supports generated to vigorously inside hollowed parts, I came up with my own pipeline:
1. First I do all the supports on full parts. 
2. I export "with supports" an STL file.
3. I import it back as a new model (with buildin supports)
4. I do my hollowing.

Unfortunately, I discovered something a hard way (2 horribly failed prints) that there is a strange bug, that sometimes, when I press "hollow" such a part - I get a model with half of the support faces with flipped normals.

See Part before "hollowing".
[image]

After "hollowing":
[image]

See how some faces got dark? That's because they are inverted.

Here is a view from meshmixer:
[image]

I ignored that fact at first, and only now discovered, that layers generated by Chitubox were freaking out in places with these inverted faces:
[image]

#### Project File (.3MF) where problem occurs
[steam-dragon-left-hand-with-support.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5141442/steam-dragon-left-hand-with-support.3mf.zip)

Also, here is a video with narration showing the bug:
https://youtu.be/k2Ws_j734ys

soswow, 2020-08-28:
Ok. I have a clue. This model when hollowed contains two separate hollow voids (not connected). I've added a hole in between two (so it connects two void, not void to outside). MAybe that's the problem. Other models where I got these flipping - might had similar situation.

soswow, 2020-08-29:
Maybe there is still something to this bug, but I think I found an improve pipeline. Not sure why I didn't do it from the start. But if I deal with support first, and then go to hollowing - it will not generate all these pesky internal supports - and all good.

bubnikv, 2021-09-22:
You came up with quite a bizzare scenario. Why are you doing it?

The support trees generated into STL are only designed to be sliced. They are made of multiple intersecting sticks. The slicing algorithm handles it correctly.

bubnikv, 2021-09-22:
Do you still get the inverted triangles around the support sticks after the hollowing? I see you tested with some alpha version that you compiled yourself. Do you get the issue with PrusaSlicer 2.4.0-alpha1?

closed 2026-09-29 by prusaslicer-bot[bot]