prusa3d/PrusaSlicer #4678: 70% of the times it doesn't let me make a hole in the hollowed model [SLA]

soswow, 2020-08-28:
### Version
2.3.0-alpha0

### Operating system type + version
MacOS

### Behavior
It tells me "Drilling holes into the mesh failed. This is usually caused by broken model. Try to fix it first." which is a lie. ... I mean ... 😩 
I mean, MAYBE it is broken. But it is a result of exporting from PrusaSlicer itself.

#### Project File (.3MF) where problem occurs
[steam-dragon-left-hand-with-support.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5141516/steam-dragon-left-hand-with-support.3mf.zip)

Video of me demonstrating a problem is here:
https://youtu.be/lqnNBQbMlPQ

soswow, 2020-08-29:
Maybe there is still something to this bug, but I think I found an improve pipeline. Not sure why I didn't do it from the start. But if I deal with support first, and then go to hollowing - it will not generate all these pesky internal supports - and all good. In that case I rearly get this problem.

bubnikv, 2021-09-22:
Hole drilling is a complex topic. We use CGAL 
https://www.cgal.org/
which is a large and for non-open source project very expensive library. It is robust, however slow and memory intensive. It expects the input meshes to be valid manifolds without cracks.

Your scenario of importing object with SLA supports as an input mesh is unsupported from us.

We have improved hole drilling in PrusaSlicer 2.4.0-alpha1, so that a hole is most likely drilled by CGAL if the triangles that the hole intersects are valid. We cannot do much better without actually fixing your mesh.

closed 2021-09-22 by bubnikv