prusa3d/PrusaSlicer #4679: Background color in settings too light

Hoop2125, 2020-08-28:
Is there a way to set the background color in the settings to a darker color? I can't read the text

[image]

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-09-01:
@Hoop2125 We need to know what macOS and PrusaSlicer version you are running and what is your settings regarding dark mode (e.g. if you're using automatic light/dark mode switching in latest macOS or similar).

The issue you're seeing appears because PrusaSlicer thinks it runs in dark mode, when it in fact isn't. That's why the text is this color. See also https://github.com/prusa3d/PrusaSlicer/issues/3697 for more details.

Hoop2125, 2020-09-01:
Prusa 2.2.0 an apple OS 10.15.6
I did go into settings/general and change to dark mode. With this setting I
can see everything in Slicer.  On Auto and light it makes the Prusa Alicer
settings impossible to read. Thanks for your help.

Michael Hooper
Hooper Printing
(616) 897-6719

FidelCapo, 2020-09-01:
@Hoop2125 There is a bug in version 2.2.0 that causes the text color change only after restarting the slicer application and should be fixed in the next version (2.3.0). Can you please confirm that the color will change after restarting the application? Thank you.

Hoop2125, 2020-09-11:
Mine does not change no matter what I change the appearance to in the general/settings. So I just set my computer to dark mode, and it is growing on me, I think I like it on dark mode LOL
Even if I quit Slicer, change the setting, and relaunch it is still wrong. When will you be releasing the update? 

-

Michael Hooper
Hooper Printing
2125 Bowes rd.
Lowell, MI 49331
616-897-6719
michael@hooperprinting.com <mailto:michael@hooperprinting.com>
http://www.hooperprinting.com <http://www.hooperprinting.com/>

lukasmatena, 2020-11-20:
It seems that this is still an issue in 2.3.0-alpha3. #5182 is another example.

Hoop2125, 2020-11-20:
I’m using the 2.3.0 Alpha and it is no longer an issue. I forget my workaround from before, butis seemed to be something to do with the Mac light/dark mode settings. I found the alpha by watching a Josef Prusa video <https://youtu.be/6LLK9soDW88> on YouTube. So far, this is working great for me. 

-

Michael Hooper
Hooper Printing
2125 Bowes rd.
Lowell, MI 49331
616-897-6719
michael@hooperprinting.com <mailto:michael@hooperprinting.com>
http://www.hooperprinting.com <http://www.hooperprinting.com/>

alsmithson, 2020-11-20:
It is still an issue in 2.3.0-alpha3. I am the one who posted [#5182](https://github.com/prusa3d/PrusaSlicer/issues/5182) because I missed this post (sorry). I am running OSX 10.14.6 Mohave in dark mode, but I've disabled dark mode for PrusaSlicer (defaults write com.prusa3d.slic3r/ NSRequiresAquaSystemAppearance -bool Yes) because PrusaSlicer dark mode looks really ugly to me. I hate dark mode on apps that use so many colors, and dark mode makes text hard to read when it is so tiny (like PrusaSlicer).

The colors are correct when I switch back to light mode in the OS.

It isn't just the green text, but I just saw that the lock icons go white even when dark mode is disabled as above.

bubnikv, 2020-12-07:
I don't remember the details, but I believe that the dark mode support in OSX Mohave is half baked and it is really best to be disabled. I am not sure whether it is worth investing our time into it, as the OSX Mohave is quite old.

closed 2026-09-29 by prusaslicer-bot[bot]