prusa3d/PrusaSlicer #4681: Too many bottom layers for skirt object

markus-i, 2020-08-29:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 2004

### 3D printer brand / version + firmware version (if known)
Originla Prusa i3 Mk3S MMU2s

### Behavior
I'm (trying to) print some technical parts in PC blend (all settings from Prusa), and had trouble with warping. Full height skirt didn't help, wide brim helped a bit - but you can't combine the two, as the slicer only attaches the brim to the ouotermost part, i.e. the skirt. So, for the most critical and complicated part, I decided to try to keep the bottom layer of the skirt object to extend the brim inwards to the actual print object. It's still printing, but from what I can see (the skirt is in the way...) it seems to look good (keeping my fingers crossed).

Anyway... when trying to extend that technique to another couple of parts, I noticed (you have to look closely) that the slicer creates 3 bottom layers for the skirt object, although only one bottom layer is selected in the object specific "layers and perimeters" dialog. As this is the default setting, I would guess that there is still a bug there.
[Achsaufnahme.zip](https://github.com/prusa3d/PrusaSlicer/files/5145125/Achsaufnahme.zip)

Reproduce: 
- create a print object and an enclosing sleeve object
- add both the print object and sleeve object to a slicer platter
- rotate and align both objects so that the sleeve completely wraps the print object
- using the object specific print settings, set the sleeve object to 0% infill, 1 bottom layer, 1 perimeter, 0 top layers
- slice
- zoom in to the bottom layers between the sleeve and the print object and count them

Right now, this isn't design breaking for me (maybe the 3 bottom layers are all that keep my object flat...), but it will make removing those layers harder - and it would be nice if the slicer actually did what I tell it.

#### Project File (.3MF) where problem occurs
.3MF is attached

rtyr, 2020-08-29:
You have minimal shell thickness set to 0.5mm. Set it to zero or 0.2 if you need 1 bottom layer.

[image]

markus-i, 2020-08-29:
Thanks a lot - that explains it.
Closing this issue.

closed 2020-08-29 by markus-i